rm-hull/luma.oled #2: Testing bug fix on SH1106 device

rm-hull, 2015-02-24:
Hi @xes,

I pushed some bug fixes and added some more examples (see 08317bc2f261073b57c45b5261a29ede69d202de) - can you run them against your SH1106 device to confirm the changes I made have worked OK?

There was a bug in rendering images (I think i made an assumption pixels were either 255 or 0, but 1-bit images like the pi_logo example are 0 or 1).

Anyway, there's now a `setup.py` which you could try testing against if you dont mind please.

P.s. I guess you'll neeed to edit the example files to reassign the correct device class.

xes, 2015-02-25:
Hi @rm-hull!
Thank you for the update. I have just checked the setup.py and all the new examples. 
Everything is working perfectly with the sh1106.
Here the photos:
[image]
[image]

rm-hull, 2015-02-25:
Great - thanks !

closed 2015-02-25 by rm-hull