rm-hull/luma.oled #3: ImportError: no module named device

nwalters512, 2015-02-24:
I followed your instructions to the T to get everything installed, but I'm running into a problem whenever I try to use the library. On the first line ```from oled.device import ...```, the python interpreter throws an ImportError saying that it cannot find the module named device. However, if I go to the directory where python modules are installed, I can see oled in the list, and it definitely contains a module called device. What could be the problem?

rm-hull, 2015-02-24:
Hi, firstly, try setting:

    $ export PYTHONPATH=.

and try running the examples again from the top directory.

In fact, you should probably pull again, as I've just committed some changes - there's now a `setup.py` which will install in the proper place. Have a look at the updated readme and let me know how you get on

rm-hull, 2015-02-24:
p.s. there also was a documentation bug which also got fixed. The first line of the example was wrong:

```python
from oled.device import ssd13055, sh1106
```

It should actually be:

```python
from oled.device import ssd1306, sh1106
```

nwalters512, 2015-02-24:
Just recloned the repository and reinstalled using setup.py, still no cigar. Your first suggestion did not help either. I have a feeling that this is just me being new at python but it shouldn't be this complicated :/

rm-hull, 2015-02-24:
Can you paste the entire error message you are getting, and what is the output from running:

    $ ls -l /usr/local/lib/python2.7/dist-packages/oled/

at the command line.

You're right, it shouldn't be this hard, but there are a lot of moving parts just to squirt an image down the wire - it's worth it though in the end ;)

I'll try and help best I can

nwalters512, 2015-02-24:
Here's the output:

```
-rw-r--r-- 1 root staff 7733 Feb 24 16:36 device.py
-rw-r--r-- 1 root staff 6426 Feb 24 16:36 device.pyc
-rw-r--r-- 1 root staff   26 Feb 24 16:36 __init__.py
-rw-r--r-- 1 root staff  184 Feb 24 16:36 __init__.pyc
-rw-r--r-- 1 root staff 2754 Feb 24 16:36 render.py
-rw-r--r-- 1 root staff 1417 Feb 24 16:36 render.pyc
```

Thanks so much for the help, it's fantastic to see the creators of open-source software supporting their work :)

rm-hull, 2015-02-24:
Ok, so you have the latest, looking at the file sizes

What output do you get if you run (at the bash prompt) in the top-level directory:

    $ python examples/demo.py

You should get a picture like this (if you've wired it up properly)
[image]

If not, is there any text/error reported at the bash shell?

nwalters512, 2015-02-24:
I see that image, and I found that my script works just fine if I run it from the top-level directory where I installed it (yay!). However, my understanding of installing a python package with ```sudo python setup.py install``` is that it is installed for use from any directory. Is this correct?

rm-hull, 2015-02-24:
Once installed with setup.py, the `oled` package should be available regardless of where you are in the file system, so you should be able to do something like:

    $ cd ~
    $ python
    Python 2.7.3 (default, Mar 18 2014, 05:13:23) 
    [GCC 4.6.3] on linux2
    Type "help", "copyright", "credits" or "license" for more information.
    >>> from oled.device import ssd1306
    >>>

And not get an import error.

As to your own scripts, if you nest them directories, then you probably ought run them from the top level, and not in the subdirectories. The other thing that always *still* catches me out is not having an empty `__init__.py` in package directories...

Anyway, its good news you got it working. Feel free to raise bugs or feature requests

nwalters512, 2015-02-25:
I still can't run my script from anywhere on the machine, it has to be in the directory containing the oled package. Any idea what gives?

rm-hull, 2015-02-25:
Can you post your script here, and I will have a look (else email it to me - richard@destructuring-bind.org)

closed 2015-03-05 by rm-hull