rm-hull/luma.oled #6: 96x16 support

dslokva, 2015-07-14:
Hello!

Please help with I2C SainSmart 96x16 display (0.69).
I can use your program to display out the text, but I need to shift down Y coordinate.
Text is visible if I use this parameters (example):
```
font = ImageFont.truetype('/usr/share/fonts/truetype/droid/DroidSans.ttf', 16)
draw.text((0, 36), "1234567890", font=font, fill=255)
```

But text is corrupted at bottom line (about 2 pixels is absent).
I can send photo if it needs.

Sorry for my english I'm from Kazakhstan.

rm-hull, 2015-07-16:
You would need to post a full example of the code you are trying to run. There are two different implementations which are subtly different in how they are initialised and controlled. A photo attached would also be useful.

dslokva, 2015-07-17:
Hi!
Here is the foto!

and full example code I use.
when i write (0,0) in draw function - there is nothing on the display.
It's a vertical shift as I understand.

I tried change device width and height in your code to 96x16, but in this case display shows only chaos dots.
```
from oled.device import ssd1306, sh1106
from oled.render import canvas
from PIL import ImageFont, ImageDraw

device = ssd1306(port=1, address=0x3C)  # rev.1 users set port=0

with canvas(device) as draw:
    #font = ImageFont.load_default()
    font = ImageFont.truetype('/usr/share/fonts/truetype/droid/DroidSans.ttf', 18)
    #draw.rectangle((0, 0, device.width, device.height), outline=0, fill=255)
    draw.text((0, 36), "1234567890", font=font, fill=255)

```

here is my system:
```
pi@raspberrypi ~/ssd1306/examples $ uname -a
Linux raspberrypi 4.0.7-v7+ #802 SMP PREEMPT Wed Jul 8 18:01:25 BST 2015 armv7l GNU/Linux

pi@raspberrypi ~/ssd1306/examples $ cat /etc/issue
Raspbian GNU/Linux 7 \n \l

```
[image]

rm-hull, 2015-07-28:
Do you know which driver chip the device uses? could you try using the `sh1106` driver instead?
If you could research which chipset it uses we could go though similar process to issue #1 where @xes helped implement the `sh1106` driver.

dslokva, 2015-07-29:
OK! I will try this solution.

closed 2016-03-03 by rm-hull