rm-hull/luma.oled #7: Blank Canvas

gizu, 2015-07-26:
Hello,

thank you very much for this nice piece of Software. I use it with Banana Pi and a SainSmart Display.
Since i`ve got no experience with python and programming in general, i stuck at one thing. It would be nice if you can wrote me a little example eg. the command to flush the canvas.

P.S.: Forgive my bad english, it`s not my native language. :x

rm-hull, 2015-07-28:
Did you look at the simple example: https://github.com/rm-hull/ssd1306/blob/master/examples/demo.py#L13 ?

You should be able to do something like:

```python
...
with canvas(device) as draw:
    draw.text(20, 20, "Hello", fill=255)

time.sleep(10)

with canvas(device) as draw:
    draw.text(20, 20, "World", fill=255)
```

Every time you `with canvas(...)` it creates a new canvas which will get flushed when the with block finishes.

gizu, 2015-07-31:
Argh, its so easy. Thank you very much. :-)

closed 2015-07-31 by gizu