rm-hull/luma.oled #10: OO Driver Usage

MilnerG, 2016-01-24:
Hello,

I'm attempting to use the driver using objects.
The problem I am facing is I have my own class that encapsulates calls to the driver, within this class I want to have an instance attribute for the driver and a function that updates the display using the driver instance attribute. However when running the application I get an inappropriate ioctl exception:

    File "/usr/local/lib/python2.7/dist-packages/oled/render.py", line 74, in __exit__
     self.device.display(self.image)
    File "/usr/local/lib/python2.7/dist-packages/oled/device.py", line 191, in display
     const.PAGEADDR,   0x00, self.pages-1)  # Page start/end address
    File "/usr/local/lib/python2.7/dist-packages/oled/device.py", line 73, in command
     self.bus.write_i2c_block_data(self.addr, self.cmd_mode, list(cmd))
    IOError: [Errno 25] Inappropriate ioctl for device

The only way I can get it to work is if I create the driver instance in the function rather than in the constructor and reusing it. This is of course inefficient as every time the display is updated a new ssd1306 instance is created.

    from oled.device import ssd1306
    from oled.render import canvas
    from PIL import ImageFont, ImageDraw
    
    class Display:
        def __init__(self):
            self.device = ssd1306(port=1, address=0x3C)
            self.font = ImageFont.load_default()
    
        def display(self):
            with canvas(self.device) as draw:
                draw.text((0, 0), "Hello World!", font=self.font, fill=255)

Thanks,
George

rm-hull, 2016-01-24:
From what I can tell your class should just work: I extended your test program as follows just to see how defining two `Display`s would interact:

```python
from oled.device import ssd1306
from oled.render import canvas
from PIL import ImageFont, ImageDraw

import time

class Display:
    def __init__(self):
        self.device = ssd1306(port=1, address=0x3C)
        self.font = ImageFont.load_default()

    def display(self, text):
        with canvas(self.device) as draw:
            draw.text((0, 0), text, font=self.font, fill=255)

d1 = Display()
d2 = Display()

while True:
    d1.display("Hello world")
    time.sleep(5)

    d2.display("Goodbye")
    time.sleep(5)
```

And it works for me - interleaving *hello word* and *goodbye*.

I tried it on a Raspberry Pi model B (one of the original 512Mb ones), with Debian Jessie and the latest firmware:

    $ uname -a
    Linux raspberrypi 4.1.13+ #826 PREEMPT Fri Nov 13 20:13:22 GMT 2015 armv6l GNU/Linux

Doing a search for  _python i2c inappropriate ioctl for device_ there are some relevant responses:

* http://raspberrypi.stackexchange.com/questions/28357/inappropriate-ioctl-for-device-error-since-upgrading-rasbian-firmware 
* https://www.raspberrypi.org/forums/viewtopic.php?p=675658#p675658

can you make sure that I2C is properly enabled by doing

    $ sudo raspi-config

And then advanced options

[image]

Then I2C

[image]

And 'Yes'

[image]

[image]

[image]

Try rebooting the Raspberry Pi and check the I2C driver is loaded:

    $ dmesg | grep i2c
    [    5.136489] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 400000)
    [    5.140345] i2c /dev entries driver

See how that goes & let me know.

MilnerG, 2016-01-24:
I doubled checked i2c configuration, it was all fine.
I'm using the Raspberry Pi 2
```
$ uname -a
Linux mypi 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015 armv7l GNU/Linux

 $ dmesg | grep i2c
[    3.885756] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 400000)
[    6.666780] i2c /dev entries driver
```
It would appear that the problem is actually being caused by python-daemon https://pypi.python.org/pypi/python-daemon/
Running my class using a simple while loop worked fine.

This is a closer implementation of what I'm trying to achieve and demonstrates the issue.
```
from oled.device import ssd1306
from oled.render import canvas
from PIL import ImageFont, ImageDraw

class Display:
    def __init__(self):
        #self.device = ssd1306(port=1, address=0x3C) #Doesn't work
        self.font = ImageFont.load_default()

    def display(self, text):
        device = ssd1306(port=1, address=0x3C) #Works
        with canvas(device) as draw: #Works
        #with canvas(self.device) as draw: #Doesn't work
            draw.text((0, 0), text, font=self.font, fill=255)

from time import sleep
from daemon import runner

class App:
    def __init__(self):
        self.stdin_path = '/dev/null'
        self.stdout_path = '/dev/tty'
        self.stderr_path = '/dev/tty'
        self.pidfile_path = '/var/run/myapp.pid'
        self.pidfile_timeout = 5
        self.display = Display()
    def run(self):
        test = 1
        while True:
            self.display.display("Test:" + str(test))
            test+=1
            sleep(5)

myapp = App()
daemon_runner = runner.DaemonRunner(myapp)
daemon_runner.do_action()
```
So this issue isn't directly related to the driver and has got something to do with how python-daemon works.

rm-hull, 2016-01-24:
If you're wanting to write a service, maybe the simplest thing would be to just write it as a regular python script and let `launchd` (or `systemd` in debian Jessie) be responsible for running it in the background.

Looking at the docs for that package it looks like you should perhaps be wrapping the program in a daemon context instead:

    with daemon.DaemonContext():
        App().run()

MilnerG, 2016-01-24:
Indeed python-daemon is poorly documented, the runner class acts as a wrapper to the daemon context providing start, stop functionality. However using the context directly seems to work and I am able to use my display class. I will look into gaining start stop functionality using systemd instead.
```
from oled.device import ssd1306
from oled.render import canvas
from PIL import ImageFont, ImageDraw

class Display:
    def __init__(self):
        self.device = ssd1306(port=1, address=0x3C)
        self.font = ImageFont.load_default()

    def display(self, text):
        with canvas(self.device) as draw:
            draw.text((0, 0), text, font=self.font, fill=255)

from time import sleep
import daemon
import lockfile

class App:
    def __init__(self):
        self.display = Display()
    def run(self):
        test = 1
        while True:
            self.display.display("Test:" + str(test))
            test+=1
            sleep(5)

context = daemon.DaemonContext(
        pidfile=lockfile.FileLock('/var/run/test.pid')
        )
with context:
    App().run()
``` 
The working code using python-daemon.

Thanks for the help.

closed 2016-01-24 by MilnerG