rm-hull/luma.oled #11: Error while using custom code

LuisDiazUgena, 2016-03-03:
Hi!

I'm trying to use your library to show the temp raspi2 and I made my [custom code](https://github.com/LuisDiazUgena/raspberry-pi-2/blob/master/oled/cpuTemp_oled.py) to do it... and it doesn't work.

It throws a traceback:
`Traceback (most recent call last):
  File "cpuTemp_oled.py", line 35, in <module>
    draw.text((margin,margin),cpuTemp,font=fontName,fill=255)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 271, in text
    mask = font.getmask(text, self.fontmode)
SystemError: error return without exception set`

I've tried the examples in the library and it works like a charm (Except for sys info, that as far as I know, needs to have another dependences installed.

What can I do?

rm-hull, 2016-03-03:
Hi Luis,

A couple of things spring to mind:

* At line 35 https://github.com/LuisDiazUgena/raspberry-pi-2/blob/master/oled/cpuTemp_oled.py#L35 I would wrap cpuTemp to cast it as a string, something like:

```python
draw.text((margin,margin),str(cpuTemp),font=fontName,fill=255)
```

* In the try/except block here https://github.com/LuisDiazUgena/raspberry-pi-2/blob/master/oled/cpuTemp_oled.py#L22-L26, it is implicitly returning a `None`, so you probably want to guard against that with as follows:

```python
cpuTemp = getTempCPU()
if cpuTemp:
    draw.text((margin,margin),str(cpuTemp),font=fontName,fill=255)
```

* The draw context only flushes when the with-block exits, so you need to invert the with and the while statements ... your code should probably look like:

```python
margin = 2
sleep_time = 2.5 #seconds

while True:
    with canvas(device) as draw:
        cpuTemp = getTempCPU()
        if cpuTemp:
            draw.text((margin,margin),cpuTemp,font=fontName,fill=255)

    sleep(sleep_time)
```

rm-hull, 2016-03-03:
p.s. To get the sysutil example running, there are some dependencies you need to install - see https://github.com/rm-hull/ssd1306/blob/master/examples/sys_info.py#L3-L7

LuisDiazUgena, 2016-03-04:
Thanks rm-hull, I've implemented the bug fixes that you suggested. Will try it when I get home and let you know. Thanks!

LuisDiazUgena, 2016-03-05:
Hi! It works! 

Thanks for your help!

closed 2016-03-05 by LuisDiazUgena