rm-hull/luma.oled #17: Publish to Pypi & add requirements.txt

rm-hull, 2016-09-24:


thijstriemstra, 2016-10-04:
requirements.txt shouldn't be necessary if you define the dependencies in setup.py..

rm-hull, 2016-10-23:
Will try to do this in the next few days

rm-hull, 2016-11-13:
Hi @thijstriemstra, I just published SSD1306 (0.3.0) to pypi for the first time (see https://pypi.python.org/pypi/ssd1306/0.3.0) .. do you know why the README.rst is not rendering?

thijstriemstra, 2016-11-13:
@rm-hull hmm that's odd.. I'll look into it.

thijstriemstra, 2016-11-13:
@rm-hull there's a util called `rstcheck` that validates rst documents.

```
pip install rstcheck
```
And that gives an error for the README:

```
$ rstcheck README.rst 
README.rst:2: (WARNING/2) Duplicate explicit target name: "datasheet".
```
It's very likely that Pypi chokes on this. I'll create a PR with a fix. Note that there's a Pypi test site as well that you can use to preview the package documentation and usage.

thijstriemstra, 2016-11-13:
Having a travis task that checks for these doc mistakes would be nice as well. I'll see if I can add one. Also a good starting point for introducing tests at some point...

rm-hull, 2016-11-13:
yay! https://testpypi.python.org/pypi?:action=display&name=ssd1306&version=0.3.1

travis task would be 👍 
test could be interesting... I guess a mock `SMBus` object would be all that is needed

thijstriemstra, 2016-11-13:
@rm-hull this looks promising: https://github.com/TCAllen07/raspi-device-mocks

thijstriemstra, 2016-11-13:
Also just noticed that the distributed tar.gz file does not contain the examples, docs, license etc.

```
$ ls -l ssd1306-0.3.1/
-rw-r--r--  1 foo  wheel  11595 13 nov 23:25 PKG-INFO
drwxr-xr-x  5 foo  wheel    170 13 nov 23:25 oled
-rw-r--r--  1 foo  wheel   1009 13 nov 23:24 setup.py
```
 I think it would be good to include these assets since the average rpi user would get confused by this IMO. I'll include it in the upcoming PR.

thijstriemstra, 2016-11-13:
Ok, the task is an improvement but it doesnt run the docs task yet under travis, will look into that.

thijstriemstra, 2016-11-14:
I see https://pypi.python.org/pypi/ssd1306/0.3.1 but when I pip install it, it's still using 0.3 (and failing with the missing README.rst error).

rm-hull, 2016-11-14:
Ok, 0.3.2 just published, and I was able to pip install it

closed 2016-11-15 by rm-hull