rm-hull/luma.oled #20: No module named oled.devie

RobodroidAnnecy, 2016-10-19:
Hi, I'm trying to use your code to run my OLED 0.96" display. However, when I run the demo.py I got this error : 
`Traceback (most recent call last):
File "demo.py", line 6, in <module>
   from demo_opts import device
File "/home/pi/ssd1306/examples/demo_opts.py", line 31, in <module>
   import oled.device
ImportError: No module named oled.device`

Do you know what should I do for that ? I've juste cloned your code to my pi in /home/pi/ssd1306 and made no modification.

thanks in advance.

Victor

rm-hull, 2016-10-19:
Hi Victor, did you follow all the setup steps here: 

* https://github.com/rm-hull/ssd1306#pre-requisites, and
* https://github.com/rm-hull/ssd1306#installing-the-python-package

From the error you describe, it sounds very much like you missed the bit about installing the library.

thijstriemstra, 2016-10-23:
Make sure to run `sudo python setup.py install` in ` /home/pi/ssd1306` after cloning.

nickehallgren, 2016-10-27:
Hi,

I noticed the same on a Raspberry pi 3 see the output:

```
pi@raspberrypi:~/ssd1306 $ sudo python setup.py install
running install
running build
running build_py
running install_lib
creating /usr/local/lib/python2.7/dist-packages/oled
copying build/lib.linux-armv7l-2.7/oled/__init__.py -> /usr/local/lib/python2.7/dist-packages/oled
copying build/lib.linux-armv7l-2.7/oled/device.py -> /usr/local/lib/python2.7/dist-packages/oled
copying build/lib.linux-armv7l-2.7/oled/render.py -> /usr/local/lib/python2.7/dist-packages/oled
byte-compiling /usr/local/lib/python2.7/dist-packages/oled/__init__.py to __init__.pyc
byte-compiling /usr/local/lib/python2.7/dist-packages/oled/device.py to device.pyc
byte-compiling /usr/local/lib/python2.7/dist-packages/oled/render.py to render.pyc
running install_egg_info
Writing /usr/local/lib/python2.7/dist-packages/ssd1306-0.2.0.egg-info
```
Installation ok

```
pi@raspberrypi:~/ssd1306 $ cd ..
pi@raspberrypi:~ $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import oled
>>> quit()
```
Works in a folder below

```
pi@raspberrypi:~ $ cd intercom/
pi@raspberrypi:~/intercom $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import oled
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "oled.py", line 1, in <module>
    from oled.device import ssd1306
ImportError: No module named device
>>>
```
but not in a folder next to the source files...

```
pi@raspberrypi:/usr/local/lib/python2.7/dist-packages $ ls -la
total 188
drwxrwsr-x 3 root staff  4096 Oct 27 14:35 .
drwxrwsr-x 4 root staff  4096 Sep 23 05:35 ..
-rw-r--r-- 1 root staff 10129 Oct 27 09:28 Adafruit_ADS1x15-1.0.2-py2.7.egg
-rw-r--r-- 1 root root  76786 Oct 25 13:24 Adafruit_GPIO-1.0.1-py2.7.egg
-rw-r--r-- 1 root root   9388 Oct 25 13:24 Adafruit_PureIO-0.2.1-py2.7.egg
-rw-r--r-- 1 root staff  9300 Oct 27 14:03 Adafruit_SSD1306-1.6.1-py2.7.egg
-rw-r--r-- 1 root staff   381 Oct 27 09:28 easy-install.pth
-rw-r--r-- 1 root staff 45512 Oct 25 11:17 keyboard-0.7.1-py2.7.egg
drwxr-sr-x 2 root staff  4096 Oct 27 14:35 oled
-rw-r--r-- 1 root staff 11472 Oct 27 14:35 ssd1306-0.2.0.egg-info
pi@raspberrypi:/usr/local/lib/python2.7/dist-packages $ cd oled/
pi@raspberrypi:/usr/local/lib/python2.7/dist-packages/oled $ ls -la
total 40
drwxr-sr-x 2 root staff 4096 Oct 27 14:35 .
drwxrwsr-x 3 root staff 4096 Oct 27 14:35 ..
-rw-r--r-- 1 root staff 7794 Oct 25 13:36 device.py
-rw-r--r-- 1 root staff 6497 Oct 27 14:35 device.pyc
-rw-r--r-- 1 root staff   50 Oct 25 13:36 __init__.py
-rw-r--r-- 1 root staff  220 Oct 27 14:35 __init__.pyc
-rw-r--r-- 1 root staff 2754 Oct 25 13:36 render.py
-rw-r--r-- 1 root staff 1417 Oct 27 14:35 render.pyc

```

and the files seems to be in the right place, any ideas?

thijstriemstra, 2016-11-02:
the title of this ticket contains a typo that is confusing, can someone fix that?

rm-hull, 2016-11-13:
@nickehallgren - in the snippet you supplied (as below), do you have a file called _oled.py_ in the intercom directory?

```
pi@raspberrypi:~ $ cd intercom/
pi@raspberrypi:~/intercom $ python
Python 2.7.9 (default, Sep 17 2016, 20:26:04)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import oled
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "oled.py", line 1, in <module>
    from oled.device import ssd1306
ImportError: No module named device
>>>
```

nickehallgren, 2016-11-15:
@rm-hull When I read your answer it hit me, feeling so stupid. Yes, my script was named oled.py. I now renamed it and everything works perfectly. Thank you!

rm-hull, 2016-11-15:
@nickehallgren - No worries, glad we were able to resolve the issue :)

@RobodroidAnnecy - are you still experiencing issues?  Did you follow the setup instructions?

rm-hull, 2016-11-23:
closing ~> no response

closed 2016-11-23 by rm-hull