rm-hull/luma.oled #21: sys_info.py - ImportError: The _imagingft C module is not installed

Xevailo, 2016-10-25:
I am using minibian on a RPi3 and followed your instruction until the installation of pillow failed. So I used
`apt-get install libjpeg-dev zlib1g-dev` and tried the installation of pillow again with `pip3 install pillow`, this time with success as `python3 demo.py` and `python3 maze.py` both worked fine. Trying to run `sys_info.py` however yielded the following result:

`Traceback (most recent call last):
  File "sys_info.py", line 77, in <module>
    main()
  File "sys_info.py", line 74, in main
    stats(oled)
  File "sys_info.py", line 65, in stats
    font2 = ImageFont.truetype('../fonts/C&C Red Alert [INET].ttf', 12)
  File "/usr/local/lib/python3.4/dist-packages/PIL/ImageFont.py", line 239, in truetype
    return FreeTypeFont(font, size, index, encoding)
  File "/usr/local/lib/python3.4/dist-packages/PIL/ImageFont.py", line 128, in __init__
    self.font = core.getfont(font, size, index, encoding)
  File "/usr/local/lib/python3.4/dist-packages/PIL/ImageFont.py", line 37, in __getattr__
    raise ImportError("The _imagingft C module is not installed")
ImportError: The _imagingft C module is not installed`

Xevailo, 2016-10-27:
Solved it:
changed the font2 declared in line 65 ti the default font.

closed 2016-10-27 by Xevailo

rm-hull, 2016-10-27:
Sorry I didn't get chance to investigate, I've been away with work related business.

rm-hull, 2016-10-27:
It sounds like you now need to install `libfreetype-dev` before installing Pillow in order to support truetype fonts (this wasn't true with the specific version of Pillow I was using) - I added a note to this effect the README with this commit: 5578f69b979b54838c2ac4a08f123a001c1d63fa

You might need to do:

    $ sudo apt-get install libfreetype-dev
    $ sudo pip --force-reinstall pillow

(or use `pip3` if you are on Python 3)

Xevailo, 2016-10-27:
Thanks for your help so far. Using `apt-get install libfreetype-dev` got me the error `Unable to locate package libfreetype-dev`. However, using 'apt-get install libfreetype6-dev` instead and then `pip install --upgrade --no-deps --force-reinstall pillow` did the trick and the font has now successfully been loaded. Thank you very much for both the program and your help!

rm-hull, 2016-10-27:
Great stuff - thanks for the taking the time to submit an issue & feedback