rm-hull/luma.oled #23: Skewed text

thijstriemstra, 2016-10-28:
I'm testing with a [128x32 OLED](http://www.ebay.com/itm/1PCS-IIC-I2C-0-91-034-128x32-white-OLED-LCD-Display-Module-3-3v-5v-For-Arduino-vv-/272362941188) and when running the sys_info demo for example, the text is unreadable. The logo example shows a stretched logo. Is there possibly an error in the width/height calculation of the ssd1306 device?

rm-hull, 2016-10-28:
It is possible - maybe quite likely. I have only ever used it on a 128x64 display against the SSD1306 IC, whereupon I got remote help from @xes to get it working on the SH1106 device (see #1).

Is the device you have definitely an SSD1306 ... can you try the SH1106 driver instead (it is in the same module). 

Uploaded photo's would help diagnose the problem further.

thijstriemstra, 2016-11-07:
I tried SH1106 and it didn't make a difference. It's horrible quality but here's a picture with the sys_info demo (without the 4th wlan line).

[image]

rm-hull, 2016-11-07:
It looks as though it is showing every other line of pixels, as if the memory mapping is interleaved.

As an experiment, can you try setting the height to `64`, and run either the *demo.py* or *maze.py* examples - these are full screen and might provide further clues as to what is happening.

thijstriemstra, 2016-11-13:
@rm-hull I can sponsor and send you one from ebay, if you can email a shipping address to my committers email address.. 👍

rm-hull, 2016-11-19:
It looks as though the multiplex settings need to be dependent on the height when the class is initialising. See conversation here: http://forum.espruino.com/conversations/269330/

rm-hull, 2016-11-27:
Related issue: #6

rm-hull, 2016-12-07:
Have a branch **feature/init_sequence** which _might_ provide the correct initialization sequence for 128x32 and 96x16 sized OLED screens.

@thijstriemstra @eudoxos - could you checkout that branch and run:

    $ sudo python setup.py install
    $ python examples/demo.py --width=128 --height=32

Let me know if that works. 

_Note that some of the other demos will probably assume height=64 - will aim to fix these up as part of the push into master._

thijstriemstra, 2016-12-07:
Will do. I would recommend `pip install -e .` from the root instead of python setup.py install..

thijstriemstra, 2016-12-07:
@rm-hull it seems everything is still squashed but only the bottom 50% is visible.

rm-hull, 2016-12-07:
Pic?

thijstriemstra, 2016-12-08:
Or it didn't happen..? ;) Let me connect pi camera.

rm-hull, 2016-12-08:
Actually .... just pushed another commit, if you could pull, install, try again pls?

thijstriemstra, 2016-12-08:
Ok first the pic; 
[image]

rm-hull, 2016-12-08:
Woudl expect to see:
[image]

thijstriemstra, 2016-12-08:
But your commit fixed it, nice!! New pic: 
[image]

The final part of the demo with the scrolling date/time is not fully visible because of height.

rm-hull, 2016-12-08:
:+1:

will fix up the sizes for the other demos in the next day or so & push it into master.

thijstriemstra, 2016-12-08:
The bounce demo goes nuts, not enough space for these balls..

eudoxos, 2016-12-08:
I tried the init_sequence branch now and I still see random pixels, though with a different pattern. The previous screen (see #70) was and still is with the master branch like this:

[image]

Now it is like this:

[image]

Sorry for bad news. Any further diagnostics I can try?

rm-hull, 2016-12-08:
@eudoxos Ok, thanks for the update: lets re-open your original issue, and move the discussion to there instead

closed 2016-12-08 by rm-hull

rm-hull, 2016-12-20:
@thijstriemstra:

> I can sponsor and send you one from ebay, if you can email a shipping address to my committers email address.. :+1: (update: on it's way)

Arrived today, and works great ... thankyou 

_[now I should go back and make sure all the demos work on that screen size]_

thijstriemstra, 2016-12-20:
Nice!

commit 5a92437d (2016-12-07): Add support for different screen sizes (SSD1306 only) - fixes #23

Note: the SSD1306 chipset is only supported presently - should there be
a need for SH1106 support, that will be addressed later.
commit 5fb65ffb (2016-12-08): Add support for different screen sizes (SSD1306 only) - fixes #23

Note: the SSD1306 chipset is only supported presently - should there be
a need for SH1106 support, that will be addressed later.
commit 6b13921a (2016-12-08): Add support for different screen sizes (SSD1306 only) - fixes #23

Note: the SSD1306 chipset is only supported presently - should there be
a need for SH1106 support, that will be addressed later.
commit 1b55f365 (2016-12-08): Merge pull request #81 from rm-hull/feature/init_sequence

Add support for different screen sizes (SSD1306 only) - fixes #23