rm-hull/luma.oled #24: Ability to Clear Display Before Drawing New Text

556duckvader, 2016-10-31:
I have one of the 4 pin 1.3" OLED(SS1106) and I am trying to find out if there is a way to clear the display before writing text to it.  I may be blind but I am just not seeing a way to do that.

rm-hull, 2016-10-31:
You're right, there is no explicit way, but how about:

```python
with canvas(device) as draw:
    # Clear screen
    draw.rectangle((0, 0, device.width-1, device.height-1), outline=0, fill=0)
    # Draw text
    draw.text((x, top),    'Hello',  font=font, fill=255)
    draw.text((x, top+20), 'World!', font=font, fill=255)
```

thijstriemstra, 2016-11-07:
Yea, I was also looking for a way to clear the display. I ended up drawing a black square. Maybe something like `clear_display(colour)` that defaults to black?

rm-hull, 2016-11-07:
The intention was never to introduce unnecessary drawing primitives, but let that facility be fulfilled by Pillow's `ImageDraw` methods. Strictly speaking a `clear_display` function should not be necessary, as the canvas buffer was always intended to be used as follows:

```python
counter = 0
while True:
    time.sleep(1)
    counter = counter + 1
    with canvas(device) as draw:
        draw.text((0,0), str(counter), font=font, fill=255)
```

For every loop when the counter is incremented, any operations inside the `with canvas(...` block would be drawn on a fresh (blank) canvas, and only on exit from the with-block would the resultant drawing be flushed to the device.

Complex layout should group all related draw methods under the same with-block so they get applied to the canvas as a whole.

closed 2016-11-13 by rm-hull

rm-hull, 2016-11-13:
I have added some new methods, and published a new 0.3.0 version:
* `device.clear()` - clears to display immediately
* `device.hide()` - put the device into low-power sleep, switching the screen off
* `device.show()` - wake the device from low-power sleep, which restores the previously displayed value

The _demo.py_ script has been updated to test these new features.