rm-hull/luma.oled #25: smbus OSError when device is not connected

thijstriemstra, 2016-11-02:
When you try to create a new device, and it's not connected, you'll get a cryptic traceback, e.g.:

```
File ":/oled/device.py", line 179, in __init__
File ":/oled/device.py", line 73, in command
File ":/smbus/util.py", line 59, in validator
File ":/smbus/smbus.py", line 275, in write_i2c_block_data
OSError: 5
```

Are you seeing this as well? Can a less cryptic, more usable error be raised here?

rm-hull, 2016-11-08:
Now following traceback is logged:

```
$ python examples/demo.py 
Traceback (most recent call last):
  File "examples/demo.py", line 6, in <module>
    from demo_opts import device
  File "/home/pi/ssd1306/examples/demo_opts.py", line 33, in <module>
    device = Device(port=args.port, address=args.address)
  File "/usr/local/lib/python2.7/dist-packages/oled/device.py", line 185, in __init__
    raise IOError(e.errno, "Failed to initialize SSD1306 display driver")
IOError: [Errno 5] Failed to initialize SSD1306 display driver
```

closed 2016-11-08 by rm-hull

thijstriemstra, 2016-11-24:
fyi with the switch to smbus2 this changed and I'm seeing a different error with master:

```
Traceback (most recent call last):
  File "/home/thijs/projects/foo/raspi-build/src/ssd1306/oled/device.py", line 198, in __init__
    super(ssd1306, self).__init__(bus, port, address)
  File "/home/thijs/projects/foo/raspi-build/src/ssd1306/oled/device.py", line 66, in __init__
    self.bus = bus or smbus2.SMBus(port)
  File "/home/thijs/.virtualenvs/foo/lib/python3.5/site-packages/smbus2/smbus2.py", line 132, in __init__
    self.open(bus)
  File "/home/thijs/.virtualenvs/foo/lib/python3.5/site-packages/smbus2/smbus2.py", line 142, in open
    self.fd = os.open("/dev/i2c-{}".format(bus), os.O_RDWR)
PermissionError: [Errno 13] Permission denied: '/dev/i2c-1'
```

rm-hull, 2016-11-24:
I would expect to see that error if the current user is not a member of the i2c group. What is the output of running `id` at a bash prompt. This is what I get:

    $ id
    uid=1001(rhu) gid=1001(rhu) groups=1001(rhu),998(i2c),999(spi)

You would need to run:

    $ sudo adduser <YOUR_USER_ID> i2c

and logout and back in again (the group membership wont take effect until you do)

thijstriemstra, 2016-11-24:
@rm-hull yea it happens on my local Ubuntu machine (that doesn't have any i2c devices but smbus2 is pip installed). I'm fine with this permission error (compared to the original error in this ticket) or the IOError you later added (and that doesn't have any effect right now).

commit ac697e9a (2016-11-08): Catch & rethrow IOErrors - fixes #25