rm-hull/luma.oled #26: Weird image after initializing ssd1306

SniperCrowbar, 2016-11-13:
Hi everyone,

i'm getting pixel everywhere after initializing with this code:
`from oled.device import ssd1306
from oled.render import canvas
from PIL import ImageFont, ImageDraw

device = ssd1306(port=1, address=0x3C)`

After using one of the examples i'm getting the same image...
I'm afraid my ssd1306 is broken :(

Hope some could help me.
[image]

rm-hull, 2016-11-13:
I think the memory buffer is uninitialised on startup - I see the same thing.
If you run:

    python examples/demo.py

It should show some geometric shapes and "hello world" ... Do you see this?

I will look at initializing the memory on startup so you won't see this in the future

SniperCrowbar, 2016-11-13:
Aaah now it works perfect!
I thought i did this before and it did not work, but now it did.

Thanks alot! :)
Image of proof:
[image]

rm-hull, 2016-11-13:
Great stuff !

rm-hull, 2016-11-13:
I just published some changes (now at v0.3.0) which should not exhibit the random dot pattern on startup, if you want to try and pull and install the new version

SniperCrowbar, 2016-11-15:
Hi, now it wont work again :(

I used this code to clear the display:
`from oled.device import ssd1306
from oled.render import canvas
from PIL import ImageFont, ImageDraw

device = ssd1306(port=1, address=0x3C)  # rev.1 users set port=0

device.clear()
device.show()`

All i get are pixel everywhere like in my first image above...
I also tried your demo.py in my first version, but i can't get it working again like before...

rm-hull, 2016-11-16:
Ok, so firstly you shouldn't need to call `clear` or `show` directly - they will be done automatically when the class is initialised.

Secondly, have you got the latest version of the library (0.3.4), and did it re-install Ok? 

Can you try the following:

    $ sudo pip install --upgrade ssd1306

 and paste the output here.

If that shows as 0.3.4, try running the demo again and let me know how you get on.

SniperCrowbar, 2016-11-16:
After complete re-install of lib 0.3.4 following output from pip upgrade:

pi@raspberrypi:~/ssd1306 $ sudo pip3 install --upgrade ssd1306
Requirement already up-to-date: ssd1306 in /usr/local/lib/python3.4/dist-packages/ssd1306-0.3.4-py3.4.egg
Requirement already up-to-date: pillow in /usr/local/lib/python3.4/dist-packages (from ssd1306)
Requirement already up-to-date: smbus2 in /usr/local/lib/python3.4/dist-packages/smbus2-0.1.4-py3.4.egg (from ssd1306)
Cleaning up...

Image after testing demo.py:
[image]

Display ON/OFF seems to work while running demo.py but nothing else is changing...

rm-hull, 2016-11-21:
So can you try  running the demo against both python versions:

    $ python example/demo.py
    $ python3 example/demo.py

To see if either work as expected, then reinstall *both* the python 2 and python 3 versions of the lib again:

    $ sudo pip install --upgrade ssd1306
    $ sudo pip3 install --upgrade ssd1306

and then try running against both python versions again:

    $ python example/demo.py
    $ python3 example/demo.py

_[must admit I'm a bit stumped here]_

rm-hull, 2016-12-04:
@SniperCrowbar - can you try and remove any old versions and try installing the 1.0.0 version that was recently released?

SniperCrowbar, 2016-12-05:
I will try it later this day and give you an update if it works or not.
Thank you for your help 👍

SniperCrowbar, 2016-12-06:
Yesterday it didn't work with freshly installed debian jessie, same result like before.
Today i will make few more tests and give you an update on my results if i got the time. :)

rm-hull, 2016-12-08:
Hi @SniperCrowbar - what device are you using? is it a RPi or some other type of device - had some interesting developments in #70 which looks related to this issue.

Can you paste the output of `dmesg | grep -i i2c` into a comment here after having run the demo code please?

SniperCrowbar, 2016-12-10:
I use an RPi B+ and this is what dmesg gave me after running demo.py

pi@raspberrypi:~ $ dmesg | grep -i i2c
[    4.579529] i2c /dev entries driver
[   11.642853] bcm2708_i2c 20804000.i2c: BSC1 Controller at 0x20804000 (irq 79) (baudrate 400000)

thijstriemstra, 2017-02-02:
@SniperCrowbar can you test with the latest version; any news?

thijstriemstra, 2017-03-17:
No feedback, closing.

closed 2017-03-17 by thijstriemstra