rm-hull/luma.oled #34: Move most of README's content to Sphinx docs

thijstriemstra, 2016-11-15:
Having the documentation in the Sphinx doc directory allows you to link to and document the code. I'd suggest to minimize the documentation in the main README.rst and move (and perhaps split into multiple docs) the bulk of the information to the doc directory. This will be published on readthedocs.io (and should be linked to from the README).

rm-hull, 2016-11-18:
Am still getting my head round the rst format, but digging the code samples (vim syntax highlights them properly).

I think the README as it stands is a bit waffly and long-winded as it is - so yes, I think it would be good to split it up and push it out into doc folder and have the readthedocs page as the primary info source.

thijstriemstra, 2016-11-18:
Cool, I'll work on it.

thijstriemstra, 2016-11-21:
I created a PR that moves all the installation stuff to sphinx but keeps the developer-related stuff in the README.

PR #37 (2016-11-21): Move most of README's content to Sphinx docs
fixes #34
closed 2016-11-21 by rm-hull