rm-hull/luma.oled #36: Recommend installation from pip

thijstriemstra, 2016-11-18:
There's no mention of `pip install ssd1306` in the documentation. The `smbus2` dependency is alos not mentioned atm.

rm-hull, 2016-11-18:
Now that the setup.py has the [install_requires](https://github.com/rm-hull/ssd1306/blob/master/setup.py#L24) I was thinking of removing the explicit pip install for pillow: it and smbus2 will be installed automatically shouldn't they?

I will add an extra note in about installing the package from PyPi though 👍

thijstriemstra, 2016-11-18:
@rm-hull yep, it will be installed automatically. I'd add a note saying just that; any **python** deps will be installed automatically with that command. apt-get stuff is still manual and needs to be installed **prior**.

rm-hull, 2016-11-27:
Fixed in commit a1e5c7e6d6e5b11afc9b063694b9e51149af3765

closed 2016-11-27 by rm-hull