rm-hull/luma.oled #38: OLED with 7 pins

thijstriemstra, 2016-11-21:
One of these [128 x 64](http://www.ebay.com/itm/-/201688735605) displays arrived but it has 7 pins and it's not clear to me where the SCL and SCA pins are.. Any idea?

rm-hull, 2016-11-22:
I think the device you have can operate either in SPI *or* I2C mode. The ebay listing identifies it as a UG-2864HSWEG01. I did a search for that product ID and here is the data sheet: https://cdn-shop.adafruit.com/datasheets/UG-2864HSWEG01.pdf

So the pins are:
* VCC - Power
* GND - Ground
* D0 - SCLK (Serial Clock)
* D1 - also known as SDIN (Serial Data Input)
* RES - ?? ... might be Read selector
* DC - Data/Command
* CS - Chip select

If you look on the back, you can see how it is configured:
[image]

You would need to set it to I2C (or IIC) mode if it is not already setup like this - this might involve some soldering to move the resistors around over the jumper pads

Alternatively, just try plugging VCC to 3.3v, GND to 0v, D0 to GPIO3 (SCL), D1 to GPIO2 (SDA) and run the `i2cdetect` command with the relevant port to see if it reports anything

If it is configured as an SPI device, then the wiring setup is a bit more complicated, but it would run faster. Unfortunately, I never merged @rogerdahl's SPI [pull request](https://github.com/rm-hull/ssd1306/pull/4) from last year (there were issues with I2C not working in wiringPi) and now there are lots of conflicts as the two repos have diverged somewhat. His idea was to abstract away the serial interface, and just have a command/data abstraction layer: it would be good to resurrect the idea and the code.

thijstriemstra, 2016-11-22:
Thanks for the details! I tried your config and unfortunately it doesn't report anything..

```
pi@neon ~ $ ls -l /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Nov 22 02:34 /dev/i2c-1
pi@neon ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```

thijstriemstra, 2016-11-22:
Looks like you were right about the welding, got a reply from the seller:

```
For this item, ours is 7 PIN and the default interface is SPI. It supports IIC with the folowing settings.:

Tear down resistor R3 and weild it to the R1 position; 
dc, cs pins have ground connection;
res pin controls via an IO to complete reset;
Resseting method: first use the RES pin to maintain a low level of 200ms;
then use the RES pin maintain high level to complete the reset operation;
then initialize the OLED, and you can use IIC.
```

Sigh.

rm-hull, 2016-11-22:
Unless you specifically need to do a hard reset, I would just keep the RES pulled up high all the time

rogerdahl, 2016-11-22:
If I remember correctly (a somewhat big "if"), my fork added SPI support but pulled in an additional GPIO library. So we wanted to refactor back to a single GPIO library, but then things stopped because I didn't have an I2C display, so wasn't able to check the I2C path and ended up breaking it.

@rm-hull If that sounds right, then, if you refactor to use a GPIO library that supports both I2C and SPI and get the I2C part working, I'll redo the SPI part and fix up the PR.

thijstriemstra, 2016-11-22:
That sounds great. I'll hold off on the soldering and look forward potentially having SPI support in this library.

rm-hull, 2016-11-22:
@rogerdahl - I've created a branch [feature/serial_interface](https://github.com/rm-hull/ssd1306/tree/feature/serial_interface) - with a serial abstraction based on your original PR (just in a separate file) and have added you and @thijstriemstra as collaborators so you should have permssion to commit directly to the branch - whether you want to do that or submit PRs to that branch, I dont mind - but I would like to get this feature fully working & tested _before_ pushing it to master.

The library has changed a bit since you last looked at it probably (mostly around docs, setup, & tests), but the core is mostly the same. I would like to continue using **smbus2** for I2C, but it looks like **gpiozero** might be a good fit for the SPI/GPIO interfacing rather than wiringPi (see https://gpiozero.readthedocs.io/en/v1.3.1/).

It would be good to try and port some of the paragraphs out of the README in your PR too - they read very well :)

There is a job that runs on travis (see https://travis-ci.org/rm-hull/ssd1306/branches) that runs `tox`, which runs some compliance checks and a few basic regression checks. You should be able to run this on your local rig by installing it with pip, and try and make the code runnable on python 2 and python 3

thijstriemstra, 2016-11-23:
If you're going to break things; and considering how stable the library already is, I would call the next release 1.0.0.

thijstriemstra, 2016-11-24:
Let me know when I can test anything (with i2c oled or spi one).

rm-hull, 2016-11-24:
The i2c implementation is done and can be tested, and there is a skeleton spi version which will just throw not implemented errors.

Install it via

    sudo python setup.py install

And remember it isn't backwards compatible!

rm-hull, 2016-11-24:
ps. It looks like gpiozero is just a wrapper around RPi.GPIO and spidev. RPi.GPIO only works on raspberry pi, so maybe using wiringpi might be better - I think it is a bit more lenient for other devices that are pin compatible with the RPi.

thijstriemstra, 2016-11-24:
@rm-hull tested the branch with an i2c device and everything still works fine.

thijstriemstra, 2016-11-27:
-1 on wiringPi or gpiozero support, spidev and RPi.GPIO has my preference.

rm-hull, 2016-11-27:
Looking at the SPI implementation just now,

rm-hull, 2016-11-27:
I've pushed some changes to the branch to implement the SPI interface - completely untested BTW, so if you could checkout the latest from [feature/serial_interface](https://github.com/rm-hull/ssd1306/tree/feature/serial_interface) and install it by doing:

    $ sudo python setup.py install

and then try and run the `demo.py` example with:

    $ python examples/demo.py -i spi

I don't imagine (for one minute) it will work the first time, so if you get errors, please paste the traceback and attach any photos to this thread. By all means, if there are typo's or dumb errors that you can fix, then just commit onto this branch directly & ping an update here.

Also, I updated the documentation: the specific version for this branch gets built automatically on push, and is available here: http://ssd1306.readthedocs.io/en/feature-serial_interface/index.html - it would be good to proof-read and suggest amends

thijstriemstra, 2016-11-30:
@rm-hull how should I connect the SPI device? Only scl and sda or?

rm-hull, 2016-11-30:
All 7 pins, as per http://ssd1306.readthedocs.io/en/feature-serial_interface/hardware.html#spi

thijstriemstra, 2016-11-30:
Oh shit sorry I didnt notice these docs, looks great!

thijstriemstra, 2016-11-30:
@rm-hull are any of these GPIO ports configurable?

thijstriemstra, 2016-11-30:
About adding the permissions for gpio etc, doesn't the `raspberrypi-sys-mods` do the same?

rm-hull, 2016-11-30:
Obviously, you can use any of the available GND and 3v3 pins for ground/power.
The SCLK, MOSI are fixed, but you can use CE1 instead of CE0 (but would need to select device=1 when creating the serial interface)

the RST and DC are defaulted to 24 / 25 but they can be configured by choosing the relevant BCM values, see: https://github.com/rm-hull/ssd1306/blob/feature/serial_interface/oled/serial.py#L74

thijstriemstra, 2016-11-30:
@rm-hull wired up and the demo throws this error:

```
$ python demo.py -i spi
Traceback (most recent call last):
  File "demo.py", line 8, in <module>
    from demo_opts import device
  File "/home/pi/projects/pi-test/ssd1306/examples/demo_opts.py", line 47, in <module>
    serial = oled.serial.spi()
  File "/home/pi/projects/pi-test/ssd1306/oled/serial.py", line 81, in __init__
    self._spi.max_bus_speed = bus_speed_hz
AttributeError: 'SpiDev' object has no attribute 'max_bus_speed'
```

rm-hull, 2016-11-30:
Change line 81 of serial.py to:
```
self._spi.max_speed_hz = bus_speed_hz
```
reinstall, retry..

thijstriemstra, 2016-11-30:
@rm-hull that fixed it. Next error:

```
$ python demo.py -i spi
Traceback (most recent call last):
  File "demo.py", line 8, in <module>
    from demo_opts import device
  File "/home/pi/projects/pi-test/ssd1306/examples/demo_opts.py", line 47, in <module>
    serial = oled.serial.spi()
  File "/home/pi/projects/pi-test/ssd1306/oled/serial.py", line 91, in __init__
    self._gpio.write(self._bcm_RST, self._gpio.HIGH)  # Keep RESET pulled high
AttributeError: 'module' object has no attribute 'write'
```

rm-hull, 2016-11-30:
haha, told you it wouldnt work first time :)

substitute any instances of `self._gpio.write` with `self._gpio.output` - I counted three on lines 90, 96, 103 or thereabouts

thijstriemstra, 2016-11-30:
HELLO FUCKING WORLD! :dancing_men:

rm-hull, 2016-11-30:
Woo-Hoo !

thijstriemstra, 2016-11-30:
@rm-hull output from the demo:

```
python demo.py -i spi
/home/pi/projects/pi-test/ssd1306/oled/serial.py:89: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(self._bcm_DC, self._gpio.OUT)
/home/pi/projects/pi-test/ssd1306/oled/serial.py:90: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(self._bcm_RST, self._gpio.OUT)
Testing basic canvas graphics...
Testing display ON/OFF...
Testing clear display...
Testing screen updates...
```
At first sight it seems very slow, especially the screen updates. I suppose a lot more tweaking is necessary. But great stuff!

rm-hull, 2016-11-30:
Try running the **bounce.py** test - that will show the frame rate - I get about 22FPS on a RPi B2 over I2C. Also there is **perfloop.py** which measures transit time through the display call - I recall that it would paint a frame in about 14ms.

I had set the default bus speed of the SPI version to 8MHz (= 8000000) in the serial constructor - it can be cranked upto 32MHz

rm-hull, 2016-11-30:
If you could also add 

```python
        self.-gpio.setwarnings(False)                        # <---- add this line
        self._gpio.setmode(self._gpio.BCM)
        self._gpio.setup(self._bcm_DC, self._gpio.OUT)
        self._gpio.setup(self._bcm_RST, self._gpio.OUT)
        self._gpio.write(self._bcm_RST, self._gpio.HIGH)  # Keep RESET pulled high
```
 into the serial constructor, and commit those changes back to the branch?

thijstriemstra, 2016-11-30:
See prev updated comment.

thijstriemstra, 2016-11-30:
@rm-hull i cant commit from that rpi, can you commit and push the changes from this ticket and I can test it after a pull.

rm-hull, 2016-11-30:
ok, no probs, i will commit
p.s. python 3.4 was measurably slower than 2.7 on my RPi

thijstriemstra, 2016-11-30:
I'm using rpi 3.

rm-hull, 2016-11-30:
pushed fixed, if you want to retest

thijstriemstra, 2016-11-30:
ps. I'd rather not add that line in the library:

```
        self.-gpio.setwarnings(False)                        # <---- add this line
```

Or did you want to silence these warnings in the examples?

thijstriemstra, 2016-11-30:
Also, these examples shouldn't throw an error on keyboard interrupt:

```
python bounce.py -i spi
^CTraceback (most recent call last):
  File "bounce.py", line 75, in <module>
    main()
  File "bounce.py", line 66, in main
    c.text((2, 0), fps, font=font, fill="white")
  File "/home/pi/.virtualenvs/pi-test/lib/python3.4/site-packages/PIL/ImageDraw.py", line 226, in text
    ink, fill = self._getink(fill)
  File "/home/pi/.virtualenvs/pi-test/lib/python3.4/site-packages/PIL/ImageDraw.py", line 121, in _getink
    ink = ImageColor.getcolor(ink, self.mode)
  File "/home/pi/.virtualenvs/pi-test/lib/python3.4/site-packages/PIL/ImageColor.py", line 128, in getcolor
    color, alpha = getrgb(color), 255
  File "/home/pi/.virtualenvs/pi-test/lib/python3.4/site-packages/PIL/ImageColor.py", line 36, in getrgb
    rgb = colormap.get(color, None)
KeyboardInterrupt
```

thijstriemstra, 2016-11-30:
After updating, no more GPIO warnings for the demo.

rm-hull, 2016-11-30:
Right - good work, I reckon.
Im going to turn in for the night now - its late.
What FPS was the bounce reporting on your RPi3?

thijstriemstra, 2016-11-30:
@rm-hull good question, I'll check it out. Thanks!

thijstriemstra, 2016-11-30:
I'll add some fixes to the demo etc. This seems to fix the slowness in the `demo` I was talking about:

```diff
diff --git a/examples/demo.py b/examples/demo.py
index c279bb4..5595507 100644
--- a/examples/demo.py
+++ b/examples/demo.py
@@ -54,7 +54,7 @@ def main():
     time.sleep(10)
 
     print("Testing display ON/OFF...")
-    for _ in range(10):
+    for _ in range(6):
         time.sleep(0.5)
         device.hide()
 
@@ -67,12 +67,12 @@ def main():
 
     print("Testing screen updates...")
     time.sleep(2)
-    for x in range(30):
+    for x in range(40):
         with canvas(device) as draw:
             now = datetime.datetime.now()
             draw.text((x, 10), str(now.date()), fill="white")
             draw.text((10, 24), str(now.time()), fill="white")
-            time.sleep(1)
+            time.sleep(.1)
 
 
 if __name__ == "__main__":
```

thijstriemstra, 2016-11-30:
Added some more doc fixes in the branch. +1 for merging this to master.

rm-hull, 2016-12-01:
Removed the setwarnings & committed a few more changes, if you could pull, install & test again please.

thijstriemstra, 2016-12-01:
@rm-hull will do

rm-hull, 2016-12-01:
@thijstriemstra - thinking of merging into master ... you've no more commits to push on this branch?

rm-hull, 2016-12-01:
Merged SPI changes to master. Gonna close this ticket as its getting too long now. Probably won't cut a release _just_ yet: any fallout from the merge, please raise as separate issues. 

Thanks for your ongoing efforts!

closed 2016-12-01 by rm-hull