rm-hull/luma.oled #42: Use enum package for constants

thijstriemstra, 2016-11-27:
instead of re-inventing the wheel, let's port the constants in this package to the [enum](https://docs.python.org/3/library/enum.html) package (available in py>=3.4 and a backport for py2).

rm-hull, 2016-11-27:
If we do use `enum`, how do you envisage unpacking the .value numeric value?

Either we would need to do something like the following everywhere:

```python
self.command(const.const.DISPLAYOFF.value,  ...)
```

or modify the `command` method to sense whether an enum or integer is supplied and call .value appropriately.

_[I think I prefer the second option, but open to suggestions]_

thijstriemstra, 2016-11-27:
Yes the 2nd option sounds good, and then maybe add a deprecation warning if an integer is passed in, or just add a comment and remove int support at some point.

rm-hull, 2016-11-27:
The command needs to be able to take integers - see https://github.com/rm-hull/ssd1306/blob/master/oled/device.py#L132-L147:

```python
            self.command(
                const.DISPLAYOFF,
                const.MEMORYMODE,
                const.SETHIGHCOLUMN,      0xB0, 0xC8,
                const.SETLOWCOLUMN,       0x10, 0x40,
                const.SETCONTRAST,        0x7F,
                const.SETSEGMENTREMAP,
                const.NORMALDISPLAY,
                const.SETMULTIPLEX,       0x3F,
                const.DISPLAYALLON_RESUME,
                const.SETDISPLAYOFFSET,   0x00,
                const.SETDISPLAYCLOCKDIV, 0xF0,
                const.SETPRECHARGE,       0x22,
                const.SETCOMPINS,         0x12,
                const.SETVCOMDETECT,      0x20,
                const.CHARGEPUMP,         0x14)
```

The non-const values are parameters to the preceding command

thijstriemstra, 2016-12-06:
Once all the test related tickets are resolved I'll take a look at this.

thijstriemstra, 2016-12-19:
Having constants for the various display and interface types would also be useful, my app (and the demo_opts) now have them hardcoded, e.g.:

```python
if args.display in ('ssd1306', 'ssd1331', 'sh1106'):
    if args.interface not in ('i2c', 'spi'):
```

thijstriemstra, 2016-12-25:
Not being able to subclass Enum and stuff is annoying, let's skip this idea.

closed 2016-12-25 by thijstriemstra