rm-hull/luma.oled #48: Anything else to do? :)

rogerdahl, 2016-11-27:
@rm-hull @thijstriemstra Thanks for all the work you've done over the last few days! I have time to chip in now, but it seems you guys have it well in hand. If there's anything you'd like me to take a look at, just create an issue and assign it to me.

rm-hull, 2016-11-27:
Hi @rogerdahl - thanks!

Well I haven't got one of the SPI OLEDs, so if you could test the implementation mentioned in #38, and provide any feedback (or push any fixes) that would be 👍 

Beyond that, it would be good to add some tests to mock the SPI and GPIO interfaces (when it's proven working maybe) - if you look at the tests I did for the i2c/smbus version, I just created a dummy implementation that recorded the commands & data and then sanity checked it. Doubtless there's probably a simpler pythonic way to do this with mock...

Else if you have time, feel free to pick up #47 or #42 - just assign to yourself

rogerdahl, 2016-11-28:
Having trouble getting SPI working. Not sure what's going on yet. I'll check with the scope later today.

I have some other full color TFT SPI displays that presumably use other controllers, that I'm also going to use with the Pi. I like the user interface of this library, with the canvas + context manager, so would it be appropriate to add non-OLED displays to this library? If so, it might be time to start thinking of a new name for the library.

thijstriemstra, 2016-11-28:
`rpi-display`?

rm-hull, 2016-11-28:
@rogerdahl - do you mean SPI in general, or this implementation? If the latter, I had used `spidev.xfer2(...)` to send commands/data - possibly this is the wrong method to call?

I would like to add support for other small OLED screens driven by the same chip - 128x32, 96x16 particularly - there should be something arriving in the post shortly hopefully, and I've just ordered a 65K color OLED device as well (driven by an SSD1331 which looks very similar in how it is driven).

Trying to drive these in python is just about bearable (actually with faster ARM chips in later Pi's, we can get some semi-decent frame-rates), but they are small 1-bit displays - I can only think it wouldn't scale up to larger full-color displays.

Noralf Tronnes has written a suite of high-quality framebuffer drivers for various LCDs in C (see https://github.com/notro/fbtft): they're fast and built into in the RPi kernel, so there is little motivation for re-inventing the same here in Python. Maybe it would be better to focus this repo on just supporting OLED-class devices with the canvas + ctx manager idea?

rm-hull, 2016-11-28:
p.s. I am in the process of tidying up two emulator devices - see the **feature/examples** branch - one writes display output to numbered PNG files (basically a screen cap), while the other uses pygame to open a display surface - this mean the lib can run on any platform that supports SDL/pygame - for example, checkout that branch, install the code, and run (this works on linux, but I cant get it to work on macOS for some reason)

    $ python example/bounce.py -d pygame

or

    $ python example/clock.py -d capture

To generate screencaps like these:

[image]

[image]

[image]

I'll create a PR with these changes if either of you could review - that'd be great. One thing that isn't quite working is the readthedocs build - it is failing because of pygame: https://readthedocs.org/projects/ssd1306/builds/4725665/ 

Both the emulators support full-color, so when the SSD1331 arrives ...

thijstriemstra, 2016-11-28:
What about having a core package that has canvas + ctx manager and let ssd1306, max7219 etc depend on this and basically only provide the required constants etc?

thijstriemstra, 2016-11-28:
> One thing that isn't quite working is the readthedocs build - it is failing because of pygame: https://readthedocs.org/projects/ssd1306/builds/4725665/

If it fails on rtd, it'll fail on the RPi as well, e.g. [here](https://github.com/rm-hull/ssd1306/blob/feature/examples/setup.py#L24) I'd really really not like to add pygame as a dependency but make it optional. Defining it in setup.py forces the dependency.

thijstriemstra, 2016-11-28:
@rm-hull If you can make a pull request once you're happy with the branch, then we can review it more easily.

rogerdahl, 2016-11-28:
@rm-hull Good point about the scaling and thanks for the pointer to fbtft.

Looks like the SPI issue is something on my side -- can't get my old fork working either! I'll investigate.

thijstriemstra, 2016-11-30:
@rm-hull seems `capture` isn't working on this RPI:

```
$ python examples/bounce.py -d capture
Traceback (most recent call last):
  File "examples/bounce.py", line 13, in <module>
    from demo_opts import device
  File "/home/pi/projects/pi-test/ssd1306/examples/demo_opts.py", line 50, in <module>
    device = Device(serial)
  File "/home/pi/projects/pi-test/ssd1306/oled/device.py", line 271, in __init__
    self.capabilities(width, height, mode="RGB")
  File "/home/pi/projects/pi-test/ssd1306/oled/mixin.py", line 40, in capabilities
    self.bounding_box = (0, 0, self.width - 1, self.height - 1)
TypeError: unsupported operand type(s) for -: 'i2c' and 'int'
```

rm-hull, 2016-11-30:
Pushed fix onto the **feature/serial_interface** branch for this

closed 2016-12-06 by rm-hull