rm-hull/luma.oled #53: Support 3-wire SPI

thijstriemstra, 2016-11-30:
From the docs "Supporting 3-wire SPI would be trivial but has not been implemented yet (no devices to test with)."

rm-hull, 2016-12-01:
Assume you read this in the hardware section in https://ssd1306.readthedocs.io: 

> 3-wire SPI eliminates the separate Data/Command line by sending an extra bit with each byte, which causes a small amount of overhead. Supporting 3-wire SPI would be trivial but has not been implemented yet (no devices to test with).

@rogerdahl - I copied this text from your original PR... Is that paragraph accurate? I thought 3-wire was where the MOSI and MISO lines were combined into a bi-directional data-line, or have I misunderstood? Here's what [wikipedia](https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Three-wire_serial_buses) has to say on it:

> ### Three-wire serial buses
> As mentioned, one variant of SPI uses single bidirectional data line (slave out/slave in, called SISO) instead of two unidirectional ones (MOSI and MISO). This variant is restricted to a half duplex mode. It tends to be used for lower performance parts, such as small EEPROMs used only during system startup and certain sensors, and Microwire. Few SPI master controllers support this mode; although it can often be easily bit-banged in software.

I think it is unlikely that we'll ever support 3-wire SPI on the basis of the above statements (slow and rare), & the fact that data transfer is not bi-directional for these displays.

It might be worth editing that statement out of the docs though.

thijstriemstra, 2016-12-01:
> It might be worth editing that statement out of the docs though.

Will do.

thijstriemstra, 2016-12-01:
Done.

rogerdahl, 2016-12-02:
The statement is still accurate. I attempted to clear up the 3/4-wire confusion in the docs at https://ssd1306.readthedocs.io/en/latest/hardware.html#i2c-vs-spi

> The use of the terms 4-wire and 3-wire SPI are a bit confusing because, in most SPI documentation, the terms are used to describe the regular 4-wire configuration of SPI and a 3-wire mode where the input and ouput lines, MOSI and MISO, have been combined into a single line called SISO. However, in the context of these OLED controllers, 4-wire means MOSI + Data/Command and 3-wire means Data/Command sent as an extra bit over MOSI.

@rm-hull Btw, the new docs look awesome! Sorry I didn't have time to help this time around. I'll see if I can get my display working over the weekend.

rm-hull, 2017-01-13:
This ticket needs to be migrated to https://github.com/rm-hull/luma.core - don't know if that's technically possible? or just close and re-create?

closed 2017-01-25 by rm-hull