rm-hull/luma.oled #55: Failing tests

thijstriemstra, 2016-12-01:
On Ubuntu 16.04 with Python 3.5.2:

```
============================================================= FAILURES ==============================================================
_________________________________________________________ test_i2c_cleanup __________________________________________________________

    def test_i2c_cleanup():
        serial = i2c(bus=smbus, address=0x9F)
        serial.cleanup()
>       smbus.close.assert_called_once()

tests/test_serial.py:71: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <Mock name='mock.close' id='139995073637848'>, name = 'assert_called_once'

    def __getattr__(self, name):
        if name in {'_mock_methods', '_mock_unsafe'}:
            raise AttributeError(name)
        elif self._mock_methods is not None:
            if name not in self._mock_methods or name in _all_magics:
                raise AttributeError("Mock object has no attribute %r" % name)
        elif _is_magic(name):
            raise AttributeError(name)
        if not self._mock_unsafe:
            if name.startswith(('assert', 'assret')):
>               raise AttributeError(name)
E               AttributeError: assert_called_once

/usr/local/lib/python3.5/unittest/mock.py:583: AttributeError
_________________________________________________________ test_spi_cleanup __________________________________________________________

    def test_spi_cleanup():
        serial = spi(gpio=gpio, spi=spidev, port=9, device=1)
        serial.cleanup()
        verify_spi_init(9, 1)
>       spidev.close.assert_called_once()

tests/test_serial.py:109: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <Mock name='mock.close' id='139995073628592'>, name = 'assert_called_once'

    def __getattr__(self, name):
        if name in {'_mock_methods', '_mock_unsafe'}:
            raise AttributeError(name)
        elif self._mock_methods is not None:
            if name not in self._mock_methods or name in _all_magics:
                raise AttributeError("Mock object has no attribute %r" % name)
        elif _is_magic(name):
            raise AttributeError(name)
        if not self._mock_unsafe:
            if name.startswith(('assert', 'assret')):
>               raise AttributeError(name)
E               AttributeError: assert_called_once

/usr/local/lib/python3.5/unittest/mock.py:583: AttributeError
___________________________________________________________ test_display ____________________________________________________________

    def test_display():
        device = sh1106(serial)
        serial.reset_mock()
    
        recordings = []
    
        def data(data):
            recordings.append({'data': data})
    
        def command(*cmd):
            recordings.append({'command': list(cmd)})
    
        serial.command = Mock(side_effect=command)
        serial.data = Mock(side_effect=data)
    
        # Use the same drawing primitives as the demo
        with canvas(device) as draw:
            baseline_data.primitives(device, draw)
    
>       serial.data.assert_called()

tests/test_sh1106.py:35: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <Mock name='mock.data' id='139995111772848'>, name = 'assert_called'

    def __getattr__(self, name):
        if name in {'_mock_methods', '_mock_unsafe'}:
            raise AttributeError(name)
        elif self._mock_methods is not None:
            if name not in self._mock_methods or name in _all_magics:
                raise AttributeError("Mock object has no attribute %r" % name)
        elif _is_magic(name):
            raise AttributeError(name)
        if not self._mock_unsafe:
            if name.startswith(('assert', 'assret')):
>               raise AttributeError(name)
E               AttributeError: assert_called

/usr/local/lib/python3.5/unittest/mock.py:583: AttributeError
```

rm-hull, 2016-12-01:
What command you running to get that output?

thijstriemstra, 2016-12-01:
simply `tox`.

thijstriemstra, 2016-12-01:
Which is strange because tox is only supposed to run 2.7 and 3.4.

```
SKIPPED:  py34: InterpreterNotFound: python3.4
ERROR:   qa: commands failed
```

thijstriemstra, 2016-12-01:
Start of qa run:

```
qa runtests: PYTHONHASHSEED='406295941'
qa runtests: commands[0] | rstcheck README.rst
qa runtests: commands[1] | flake8
qa runtests: commands[2] | coverage run -m py.test -v
======================================================== test session starts ========================================================
platform linux -- Python 3.5.2, pytest-3.0.4, py-1.4.31, pluggy-0.4.0 -- /home/thijs/projects/foo/src/ssd1306/.tox/qa/bin/python3.5
cachedir: .cache
rootdir: /home/thijs/projects/foo/src/ssd1306, inifile: 
plugins: cov-2.4.0
collected 17 items 

tests/test_capture.py::test_noops PASSED
tests/test_capture.py::test_display PASSED
tests/test_serial.py::test_i2c_init_no_bus PASSED
tests/test_serial.py::test_i2c_init_bus_provided PASSED
tests/test_serial.py::test_i2c_command PASSED
tests/test_serial.py::test_i2c_data PASSED
tests/test_serial.py::test_i2c_data_chunked PASSED
tests/test_serial.py::test_i2c_cleanup FAILED
tests/test_serial.py::test_spi_init PASSED
tests/test_serial.py::test_spi_command PASSED
tests/test_serial.py::test_spi_data PASSED
tests/test_serial.py::test_spi_cleanup FAILED
tests/test_sh1106.py::test_display FAILED
tests/test_ssd1306.py::test_init PASSED
tests/test_ssd1306.py::test_hide PASSED
tests/test_ssd1306.py::test_show PASSED
tests/test_ssd1306.py::test_display PASSED
```

rm-hull, 2016-12-01:
hmmm ... is it worth removing the `.tox` directory?

thijstriemstra, 2016-12-01:
I did `tox -r` (which is like make clean and empties the .tox dir) but same result. Let's keep this open and I'll look at it some time later.

rm-hull, 2016-12-04:
So I don't really understand what is going on with this,
* clearly it works on Travis
* my ubuntu machine (which used to work) now fails with:

```
qa inst-nodeps: /home/rhu/dev/ssd1306/.tox/dist/ssd1306-1.1.0.zip
qa installed: alabaster==0.7.9,Babel==2.3.4,configparser==3.5.0,coverage==4.2,docutils==0.12,enum34==1.1.6,flake8==3.2.1,funcsigs==1.0.2,imagesize==0.7.1,Jinja2==2.8,MarkupSafe==0.23,mccabe==0.5.2,mock==2.0.0,pbr==1.10.0,Pillow==3.4.2,py==1.4.31,pycodestyle==2.2.0,pyflakes==1.3.0,pygame==1.9.2b8,Pygments==2.1.3,pytest==3.0.4,pytest-cov==2.4.0,pytz==2016.7,RPi.GPIO==0.6.3,rstcheck==2.2,six==1.10.0,smbus2==0.1.4,snowballstemmer==1.2.1,Sphinx==1.4.9,spidev==3.2,ssd1306==1.1.0
qa runtests: PYTHONHASHSEED='1665916285'
qa runtests: commands[0] | rstcheck README.rst
qa runtests: commands[1] | flake8
qa runtests: commands[2] | coverage run -m py.test -v
=================================================================== test session starts ===================================================================
platform linux2 -- Python 2.7.12+, pytest-3.0.4, py-1.4.31, pluggy-0.4.0 -- /home/rhu/dev/ssd1306/.tox/qa/bin/python
cachedir: .cache
rootdir: /home/rhu/dev/ssd1306, inifile: 
plugins: cov-2.4.0
collected 2 items / 3 errors 

========================================================================= ERRORS ==========================================================================
__________________________________________________________ ERROR collecting tests/test_serial.py __________________________________________________________
ImportError while importing test module '/home/rhu/dev/ssd1306/tests/test_serial.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests/test_serial.py:7: in <module>
    from mock import patch, call, Mock
E   ImportError: cannot import name patch
__________________________________________________________ ERROR collecting tests/test_sh1106.py __________________________________________________________
ImportError while importing test module '/home/rhu/dev/ssd1306/tests/test_sh1106.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests/test_sh1106.py:7: in <module>
    from mock import Mock
E   ImportError: cannot import name Mock
_________________________________________________________ ERROR collecting tests/test_ssd1306.py __________________________________________________________
ImportError while importing test module '/home/rhu/dev/ssd1306/tests/test_ssd1306.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
tests/test_ssd1306.py:7: in <module>
    from mock import call, Mock
E   ImportError: cannot import name call
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 3 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
================================================================= 3 error in 0.17 seconds =================================================================
ERROR: InvocationError: '/home/rhu/dev/ssd1306/.tox/qa/bin/coverage run -m py.test -v'
_________________________________________________________________________ summary _________________________________________________________________________
  py27: commands succeeded
SKIPPED:  py34: InterpreterNotFound: python3.4
ERROR:   qa: commands failed
```

rm-hull, 2016-12-04:
The errors I encountered were because in a previous version there was a `mock.py` class in the tests directory (which I had removed) but the `mock.pyc` remained, and this was being picked up - working for me now.

rm-hull, 2016-12-06:
re. the failing tests when you run with Python 3.5.2:

I've created a PR on branch **feature/unsafe_mocks**, can you check that out and see if it works for you & if it does, please review and merge the changes onto master.

_unittest.mock_ was added in 3.3 - should there be another travis target that runs the tests in Python 3 - looking [here](https://travis-ci.org/rm-hull/ssd1306/builds/181774708#L380), the tests are run under 2.7.9 ?

thijstriemstra, 2016-12-06:
Will test. And yes, the tests should be run on both 2.7 and 3.4 (and 3.5 which is allowed to fail if it isnt available).

thijstriemstra, 2016-12-07:
Here's Travis showing the same error on Python 3.5: https://travis-ci.org/rm-hull/ssd1306/jobs/182130845

thijstriemstra, 2016-12-08:
Fixed with #77.

commit b22d1074 (2016-12-06): Add unsafe=True to mocks - fixes #55
commit 8404ef2e (2016-12-07): Merge pull request #72 from rm-hull/feature/unsafe_mocks

Add unsafe=True to mocks - fixes #55
closed 2016-12-08 by thijstriemstra