rm-hull/luma.oled #60: Qt emulator

thijstriemstra, 2016-12-03:
Being able to produce [ImageQt](http://pillow.readthedocs.io/en/3.2.x/reference/ImageQt.html) objects and rendering and previewing them in a Qt application would be cool.

thijstriemstra, 2016-12-13:
Some thoughts/sketch/example:

```python
import os
import sys

from PyQt5 import QtWidgets

from PIL import Image
from PIL.ImageQt import ImageQt

from oled.render import canvas

from demo_opts import device

def on_click(event):
    with canvas(device) as draw:
        img_path = os.path.abspath(os.path.join(os.path.dirname(__file__),
            'images', 'pi_logo.png'))
        logo = Image.open(img_path)

        # draw to OLED
        draw.bitmap((32, 0), logo, fill='white')

        # XXX: render to Qt widget.
        # qlogo is an instance of a QtGui.QImage subclass, which means that you
        # can pass the object directly to PyQt4/5 API functions and methods.
        qlogo = ImageQt(logo)

def main():
    app = QtWidgets.QApplication(sys.argv)
    window = QtWidgets.QMainWindow()

    button = QtWidgets.QPushButton("Hello SSD1306!")
    button.clicked.connect(on_click)

    window.setCentralWidget(button)
    window.show()

    app.exec_()

if __name__ == '__main__':
    main()
```

rm-hull, 2016-12-13:
I don't know much (anything, actually) about Qt, so what do you propose: 
* create a qtwidget class that extends `oled.emulator.emulator` (and some other QT class - `QtGui.QImage` perhaps?), so that it could be embedded in another QT component
* create a qtwidget class that extends `oled.emulator.emulator`, and then a method that returns a `pillow.ImageQT` instance that can be referenced inside a another QT component

thijstriemstra, 2016-12-15:
Created a branch for this at `qt-emulator-60` and added some changes but nothing working yet. PyQt5 apps in general need at least one QApplication instance. In there you can create a (main)window and in there widgets can be added. pygame works similar but more minimalistic. For now I create the QApplication inside the demo_opts module, and later I'll add that device where the demos can write their content to. I added a `emulator` package so I could add a separate `qt` module so we can separate these PyQt5 imports and classes. tbc..

thijstriemstra, 2016-12-27:
@rm-hull one of the things I'm running into is the `while True:` loops used in the examples, making it impossible to make them async. Thoughts?

rm-hull, 2016-12-29:
Hmmm, difficult - I'd rather not make the examples async, for the sake of one emulator. The purpose of the  examples is mainly to show library users (who might well be python beginners) how to use the library.  Using asyncio adds another level of complexity, as well as excluding python 2 usage.

I'm away at the moment, but will have a think too see if there is a good compromise.

rm-hull, 2017-01-13:
This ticket and any outstanding code changes should be applied to https://github.com/rm-hull/luma.core

closed 2017-01-26 by thijstriemstra