rm-hull/luma.oled #62: Add example code to demonstrate font drawing capability

rm-hull, 2016-12-04:
After d7e1fa65abc499e45579a5b9ba3f4d4883d91a44

thijstriemstra, 2016-12-08:
What do you have in mind for this?

rm-hull, 2016-12-08:
mmm ... i _kinda_ had something like this in mind: https://www.youtube.com/watch?v=I9HAg1uFlIg - much more low res and 2D though, maybe letters from the word 'Welcome' fly in, and for each iteration the word is a different language and a different font & different size. Was going to just grab a few distinctive TTFs from a free font site.

Alternatively we could have a virtual canvas, with words in different langs/fonts pasted on them
at various places, and then tack between them with some LERPing & heavy use of the`set_position` - this is dependent on the viewport branch I'm working on with at the moment. 

Aims to demonstrate:
* Use of different fonts
* Non-English character sets
* Scrolling & animation capability

thijstriemstra, 2016-12-08:
> dependent on the viewport branch I'm working on with at the moment.

ok.

commit 6fa7eda7 (2016-12-10): Add example code to demonstrate font drawing capability - fixes #62
closed 2016-12-10 by rm-hull