rm-hull/luma.oled #63: Pip installation problem

thijstriemstra, 2016-12-05:
I installed v1.0.0 from pip and I'm seeing this error:
```
$ python
Python 3.5.2 (default, Oct 24 2016, 02:11:59) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import ssd1306
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named 'ssd1306'
>>> 
```
But it's actually installed:
```
$ pip list
...
SQLAlchemy (1.0.15)
ssd1306 (1.0.0)
tabulate (0.7.7)
...
```

```
$ pip install ssd1306
Collecting ssd1306
  Downloading ssd1306-1.0.0.tar.gz (2.3MB)
    100% |████████████████████████████████| 2.3MB 219kB/s 
Requirement already satisfied: pillow in /home/foo/.virtualenvs/bar/lib/python3.5/site-packages (from ssd1306)
Requirement already satisfied: smbus2 in /home/foo/.virtualenvs/bar/lib/python3.5/site-packages (from ssd1306)
Requirement already satisfied: spidev in /home/foo/bar/spidev (from ssd1306)
Requirement already satisfied: RPi.GPIO in /home/foo/.virtualenvs/bar/lib/python3.5/site-packages (from ssd1306)
Collecting pygame (from ssd1306)
  Downloading pygame-1.9.2rc1-cp35-cp35m-manylinux1_x86_64.whl (10.0MB)
    100% |████████████████████████████████| 10.0MB 127kB/s 
Building wheels for collected packages: ssd1306
  Running setup.py bdist_wheel for ssd1306 ... done
  Stored in directory: /home/foo/.cache/pip/wheels/05/c6/7a/5a8e80c710167ca0c3cb74ddb082521e0d085cf24c006adb3b
Successfully built ssd1306
Installing collected packages: pygame, ssd1306
Successfully installed pygame-1.9.2rc1 ssd1306-1.0.0
```
Looks like there aren't any source files installed at all:
```
$ ls -lha /home/foo/.virtualenvs/bar/lib/python3.5/site-packages/ssd1306-1.0.0.dist-info
total 44K
drwxrwxr-x   2 foo foo 4,0K dec  5 21:55 .
drwxrwxr-x 132 foo foo  12K dec  5 21:55 ..
-rw-rw-r--   1 foo foo 2,6K dec  5 21:55 DESCRIPTION.rst
-rw-rw-r--   1 foo foo    4 dec  5 21:55 INSTALLER
-rw-rw-r--   1 foo foo 3,6K dec  5 21:55 METADATA
-rw-rw-r--   1 foo foo 1,3K dec  5 21:55 metadata.json
-rw-rw-r--   1 foo foo 1,3K dec  5 21:55 RECORD
-rw-rw-r--   1 foo foo    5 dec  5 21:55 top_level.txt
-rw-rw-r--   1 foo foo   96 dec  5 21:55 WHEEL
```

thijstriemstra, 2016-12-05:
Fuck, this has bitten me before. The package is called `oled`, not `ssd1306` :(

closed 2016-12-05 by thijstriemstra

rm-hull, 2016-12-05:
:)

maybe its time to rename the project...