rm-hull/luma.oled #66: Examples issue

thijstriemstra, 2016-12-05:
Using latest master with py3.4:

```
$ python maze.py -i spi
Traceback (most recent call last):
  File "maze.py", line 9, in <module>
    from demo_opts import device
  File "/home/pi/projects/pi-test/ssd1306/examples/demo_opts.py", line 41, in <module>
    bcm_RST=args.bcm_reset)
  File "/home/pi/projects/pi-test/ssd1306/oled/serial.py", line 80, in __init__
    self._spi.open(port, device)
AttributeError: 'module' object has no attribute 'open'
```

rm-hull, 2016-12-05:
Urg! -sorry this was a regression in 4f85dbdd2739a1aae6f2bbcfba4ef4b26aed07eb

rm-hull, 2016-12-05:
Try pulling from master again pls, and if this works I'll release a new version, **1.1.0**

closed 2016-12-05 by rm-hull

thijstriemstra, 2016-12-05:
@rm-hull it works now.

commit 2538e398 (2016-12-05): Fix regression in SPI initialisation - fixes #66

Caused by 4f85dbd