rm-hull/luma.oled #68: pygame requires extra dependency

thijstriemstra, 2016-12-06:
Because pygame is a dependency an extra dependency is required, because the installation fails with:

```
Running pygame-1.9.2rc1/setup.py -q bdist_egg --dist-dir /tmp/easy_install-Lxwqwp/pygame-1.9.2rc1/egg-dist-tmp-Td5P4G

WARNING, No "Setup" File Exists, Running "config.py"
Using UNIX configuration...

Hunting dependencies...
sh: 1: sdl-config: not found
sh: 1: sdl-config: not found
sh: 1: sdl-config: not found
WARNING: "sdl-config" failed!
error: Setup script exited with Unable to run "sdl-config". Please make sure a development version of SDL is installed.
```

I'd really prefer not to define it in setup.py because I don't need it and it's causing problems like this. If it's optional than people can install these extra dependencies, inc. pygame.

thijstriemstra, 2016-12-06:
The actual package that also needs to be installed: `sudo apt-get install libsdl1.2-dev`

thijstriemstra, 2016-12-06:
But then `tox` eventually fails with:

```
WARNING, No "Setup" File Exists, Running "config.py"
Using UNIX configuration...

Hunting dependencies...
SDL     : found 1.2.15
FONT    : not found
IMAGE   : not found
MIXER   : not found
PNG     : found
JPEG    : found
SCRAP   : found
PORTMIDI: not found
PORTTIME: not found
FREETYPE: found 2.6.1

Warning, some of the pygame dependencies were not found. Pygame can still
compile and install, but games that depend on those missing dependencies
will not run. Would you like to continue the configuration? [Y/n]:y

If you get compiler errors during install, doublecheck
the compiler flags in the "Setup" file.

Continuing With "setup.py"
error: [Errno 2] No such file or directory: 'version.py.in'
```

thijstriemstra, 2016-12-06:
Also, shouldn't [this import](https://github.com/rm-hull/ssd1306/blob/master/oled/emulator.py#L38) be higher up; in the `pygame` class?

rm-hull, 2016-12-06:
What distro and arch are you on that requires SDL for pygame?
I have installed it on ubuntu (x64), macOS, debian (RPi), armbian (OrangePi) and haven't seen any of these specific issues (although, I can't claim that libsdl wasnt already installed on any of them). 

Which version of python?

 > Also, shouldn't this import be higher up; in the pygame class?

Possibly, but I originally wanted to call the emulator class `pygame` (maybe in hindsight not a good idea) and this clashed; additionally I didn't want to introduce an import unnecessarily if it wasn't going to be used.

thijstriemstra, 2016-12-06:
That's on Ubuntu 16.04 LTS with compiled Python 3.5.2.

thijstriemstra, 2016-12-06:
> Possibly, but I originally wanted to call the emulator class pygame (maybe in hindsight not a good idea) and this clashed; additionally I didn't want to introduce an import unnecessarily if it wasn't going to be used.

Because all the emulators inherit that pygame import now, even though they're not using it.

rm-hull, 2016-12-06:
ah, sorry, all the emulators _do_ actually use pygame: the scaling is done by pygame's built-in scale2x and smoothscale transforms

thijstriemstra, 2016-12-06:
Hmm, are these not available in Pillow? I'll have to take a closer look. With my suggested Qt emulator (#60 ) I was thinking it would all be done by Pillow and Qt in that case.

rm-hull, 2016-12-07:
Am seeing the same problems locally now as you.. Previously it was working for me with pygame-1.9.2b6, but seems like this new [release candidate](https://pypi.python.org/pypi/Pygame/1.9.2rc1) is causing me grief... it was released 2 days ago

Looking at https://travis-ci.org/rm-hull/ssd1306#L265 but thats OK

thijstriemstra, 2016-12-07:
Do you have mock installed? Does the code look for python 3 mock first? I thought that might be the problem but haven't tested..

rm-hull, 2016-12-09:
Pygame 1.9.2rc1 introduces some extra dependencies that aren't automatically met any more on some platforms.

--> Removed install dependency of pygame - using the emulators will now raise a runtime exception with message _Emulator requies pygame to be installed_

Fixed in commit 135afd473d07787db22caf480f74c5c24ee9af60

closed 2016-12-09 by rm-hull