rm-hull/luma.oled #70: Random pixels with ssd1306 (Adafruit 128x32)

eudoxos, 2016-12-06:
I am using the current git, running on Odroid-C2. The device is [Adafruit 931](https://www.adafruit.com/product/931). Display on/off works fine, but pixels shown are random. This is the command I use:

    python3 demo.py --dim 128x32 -d ssd1306 -i i2c --i2c-port 2 --i2c-address 60

The pixel pattern is not changing while any example runs (including those which should, I guess, be moving, such as clock.py), and actually is always the same whatever runs. I can give more diagnostics if needed, now I just think ssd1306 is writing to some other place in memory as what is shown does not change at all.

[image]

rm-hull, 2016-12-06:
Hi @eudoxos - so the fact that the display on/off works Ok means that it is delivering commands to the correct port/address, but the initialisation sequence is _currently_ programmed specifically for 128x64 sizes - I think this is related to issue #23, and (courtesy of @thijstriemstra) I should be getting a 128x32 device to test with shortly. (I've also got a 96x16 on order too)

Once the device arrives, I will alter the `capabilities` class to handle the different devices initialisation sequences, so there should be no change to the interface.

eudoxos, 2016-12-06:
You might want to look at https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py#L234 -- there are subclasses for different sizes, each with its own initialization sequence.

If you want one more 128x64, I am ready to order for you, since your lib is (will be!) really helpful :)

thijstriemstra, 2016-12-06:
Marking this as a duplicate of #23.

rm-hull, 2016-12-06:
@eudoxos:

> If you want one more 128x64, I am ready to order for you

Thats really kind - thanks for the offer, but it would be more valuable if you were able to commit some time to help diagnose & test issues in the library for your Adafruit 931 on the Odroid, and provide your feedback here, either in new issues or contribute to discourse in comments in existing issues, or better still, help with documentation & code fix PRs.

I quite like the idea of collating and maintaining a wiki page (as suggested by @thijstriemstra in #71), but more generally, I would like to move the project in the following direction:
* get a good detailed view of which OLED devices we can support on which SoCs (Raspberry Pi, Odroid, Banana Pi, etc), 
* what library dependencies are required, across different distros 
* generally improving the documentation,
* provide richer examples to showcase what the library is capable of, and
* add a troubleshooting guide

eudoxos, 2016-12-07:
I find this library conveniently platform-independent, as as long as smbus2 is ported (RPi.GPIO is out of play for Odroid).
I can try to diagnost, but need some guidance. Where do you think the difference will be? Just the initialization sequence? Or something else elsewhere? I can try the sequence from Adafruit, changing what is different in the code directly; later when you add the capabilities class, that will be used for the 128x32 displays.

rm-hull, 2016-12-07:
I have an Orange Pi, and in a similar position with RPi.GPIO unfortunately. At some point I will try and source a platform-independent GPIO library and use that instead.

If you can hang on a few days, I have an idea what needs changing and will push out a new build for you to test - and yes, the changes will most likely be in the capabilities class and will provide the correct initialization for 128x64, 128x32, 96x16 screen sizes (possibly other sizes too).

eudoxos, 2016-12-07:
FWIW this library aims at cross-platform GPIO https://github.com/adafruit/Adafruit_Python_GPIO, maybe coding a orange pi backend would be easier than starting from scratch. Also, it seems that many boards provide modifier wiringPi port (this includes Odroid and Orange - http://www.orangepi.org/Docs/WiringPi.html) though it does not seem that anyone is doing proper integration for all boards under one roof. Strange, that is potentially so useful.

eudoxos, 2016-12-08:
For the record, copying my comment to #23:

With the init_sequence branch, the display looks like this:

[image]

(the pattern is consistently identical garbage, juding by the pixel formation in the upper left corner)

rm-hull, 2016-12-08:
@eudoxos Looking at your screenshot, can you let me know how you've wired it up? I wonder if you need might need to pull up the RST line (as it looks like it is currently not connected).

Similarly 3.3V looks like it is not connected, but VIN is - is that wired to the 5V line?

eudoxos, 2016-12-08:
VIN is fed 5V, and there is built-in level shifter which takes it down to 3.3V and makes this available on the 3.3V pin for the user, if needed (I don't need it, so I leave it unconnected).

The RST pin, not sure where to connect it; I thought it was some leftover for SPI models? Looking at https://learn.adafruit.com/monochrome-oled-breakouts/wiring-128x32-i2c-display they *do* connect the RST to something... as far as I know I am only supposed to provide voltage (GND and VIN) plus 2 wires for i2c.

rm-hull, 2016-12-08:
The I2C device I have doesn't have a RST pin, but conventionally it should be pulled high for normal mode, and pulled low for a brief period to cause a reset effect. Having it floating (i.e. not connected) is probably not going to be stable. 

So you have a choice:

* either pull it up using a resistor to the +5V line (so it's in normal mode all the time), or
* connect it to pin 24 and use a GPIO library to programmatically pull it high - since you are on Odroid, this probably precludes use of RPi.GPIO (this is what the SPI version does, see [here](https://github.com/rm-hull/ssd1306/blob/master/oled/serial.py#L89-L90)).

Personally, I would be inclined to try the first option... as to the resistor value, have a look here: http://electronics.stackexchange.com/questions/23645/how-do-i-calculate-the-required-value-for-a-pull-up-resistor, but I would try 10 kΩ .... but usual caveats apply: do your research / I haven't tried this / not my fault if it blows up, etc, etc :-)

eudoxos, 2016-12-08:
I added a 10kΩ pull-up resistor as you suggested, connecting the (so far unused) 3.3V and RST pins. The display did not explode, but still shows random garbage. The on-off works just fine, as said above, so it does not feel like "unstable".

rm-hull, 2016-12-08:
ok, the other option is to look at the CHARGEPUMP and PRECHARGE values - could you try editing the _oled/device.py_ file and change the following lines in the `ssd1306.__init__` method:

```python
            self.command(
                const.DISPLAYOFF,
                const.SETDISPLAYCLOCKDIV, 0x80,
                const.SETMULTIPLEX,       0x3F,
                const.SETDISPLAYOFFSET,   0x00,
                const.SETSTARTLINE,
                const.CHARGEPUMP,         0x10,      ### CHANGE 0x14 to 0x10
                const.MEMORYMODE,         0x00,
                const.SEGREMAP,
                const.COMSCANDEC,
                const.SETCOMPINS,         0x12,
                const.SETCONTRAST,        0xCF,
                const.SETPRECHARGE,       0x22,      ### CHANGE 0xF1 to 0x22
                const.SETVCOMDETECT,      0x40,
                const.DISPLAYALLON_RESUME,
                const.NORMALDISPLAY)
```

(see https://github.com/rm-hull/ssd1306/blob/master/oled/device.py#L196-L211)

Then either re-install the library, and run:

    $ PYTHONPATH=. python examples/demo.py --width 128 --height 32 -d ssd1306 -i i2c --i2c-port 2 --i2c-address 60

If that works, then we can think about a blending a flag that allows this to be swapped in as necessary

rm-hull, 2016-12-08:
@eudoxos if you can do your testing on the **feature/init_sequence** branch

eudoxos, 2016-12-08:
I've been testing both branches and running w/o installation with `PYTHONPATH=. python3 examples/demo.py ...` from each branch dir to make sure I am using the right version. Both the same (only the pattern is different for each).

rm-hull, 2016-12-08:
Does your device work with the adafruit oled library?

eudoxos, 2016-12-08:
With chargepump 0x10 and precharge 0x22 as you suggested, the display does not even light up. I checked adafruit's sources https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py#L279 and they seem to be doing exactly the same as ssd1306 does... ?

eudoxos, 2016-12-08:
Testing adafruit oled library: never tested, not able to run it on Odroid.

eudoxos, 2016-12-08:
This is with Adafruit OLED library, running on a Raspberry Pi. So the display itself is okay.

[image]

I will now try to run with init_sequence branch on the RPi.

eudoxos, 2016-12-08:
Heh, the ssd1306 init_sequence branch works just fine on the RPi with the same device. (Funnily, the demo.py screen is still rolling as I interrupted the adafruit's demo before and the display still remembers to shift screen :) ). Eh... what now?

I connected the display the same way as on the odroid-C@ before, i.e. VIN from the 5V rail, GND, SCL and SDA connected to their respective pins, RST and 3.3V connected using the pull-up 10kΩ resistor.

rm-hull, 2016-12-08:
ooh, interesting. So it works on RPi with the same wiring, but not on Odroid ?

* What OS & kernel are you running on Odroid?
* What does running `dmesg | grep -i i2c` return after you have tried to run the demo on the Odroid?
* What is the output of `ls -l /sys/bus/i2c/devices/i2c*`
* I notice you are calling the demo with `--i2c-port=2` - how may I2C buses are there on the Odroid?
* What is the output of `i2cdetect -y 0`, `i2cdetect -y 1`, `i2cdetect -y 2`, ... ?

eudoxos, 2016-12-08:
* `uname -a`
  `Linux odroid64 3.14.65-65 #1 SMP PREEMPT Sat May 28 02:50:51 BRT 2016 aarch64 aarch64 aarch64 GNU/Linux`
* `cat /etc/issue`
  `Ubuntu 16.04 LTS \n \l`
* `dmesg|grep -i i2c`
  ```
  [    0.523273] i2c /dev entries driver
  [   10.226331] aml_i2c version: 20140813
  [   10.243067] aml-i2c c1108500.i2c: add adapter aml_i2c_adap1(ffffffc073b4f0e0)
  [   10.243326] aml-i2c c11087c0.i2c: add adapter aml_i2c_adap2(ffffffc05a5f70e0)
  ```
* `ls -l /sys/bus/i2c/devices/i2c*`
  ```
  lrwxrwxrwx 1 root root 0 Feb 11  2016 /sys/bus/i2c/devices/i2c-1 -> ../../../devices/i2c-1
  lrwxrwxrwx 1 root root 0 Feb 11  2016 /sys/bus/i2c/devices/i2c-2 -> ../../../devices/i2c-2
  ```
* There are 2 buses (both meant to be used by users; bus2 is pins 27 & 28, which RPi uses as HAT sense).
* `i2cdetect -y 1` (0x51 is the RTC add-on board)
  ```
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
  00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
  10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  50: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  70: -- -- -- -- -- -- -- --
  ```
* `i2cdetect -y 2` (0x3c is the OLED)
  ```
      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
  00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
  10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
  40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
  70: -- -- -- -- -- -- -- --
  ```

rm-hull, 2016-12-08:
I assume it is this pin header:

[image]

Have you tried connecting it to bus 1 (pins 14/16) - the OLED should be able to co-exist on the same bus as your RTC board as they are at different addresses ?

_p.s. the 128x32 code is now in master, so please use that branch from now on_

eudoxos, 2016-12-08:
The pinout for C2 is at http://www.hardkernel.com/main/products/prdt_info.php?g_code=G145457216438&tab_idx=2 (bottom):

[image]

I managed to run the Adafruit OLED library now by changing the Adafruit_GPIO to think I am at beaglebone and make all GPIO calls no-op. It shows garbage on the display :))) The GPIO calls are pulling pin 24 (GPIO) high, low and high (see https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/Adafruit_SSD1306/SSD1306.py#L151). Curiously, that GPIO was was not connected on the RPi either (unless I am mixing physical/chip/human GPIO numbers) and the display still worked...

This reset behavior reminds me of http://electronics.stackexchange.com/a/186992/58205 (but might be unrelated):

> Beware of the RES-pin. It needs a low pulse at startup and high voltage during operation (as in SPI mode). I suggest a 100nF cap to GND and a 10k res to VCC, works like a charm!

I will try on the other bus tomorrow, it is a bit complicated space-wise (ribbon cable to a juice4halt module, RTC, all mounted in a case etc).

eudoxos, 2016-12-08:
Tried on the first i2c bus of the Odroid C2. Same result. The garbage pattern is the same, both for your lib and Adafruit's :|

eudoxos, 2016-12-08:
I also posted at Adafruit's forum (https://forums.adafruit.com/viewtopic.php?f=47&t=107940) since it might be some issue on their side.

eudoxos, 2016-12-13:
I have not got answers from neither odroid nor adafruit. I am concluding there is some mysterious problem. Meanwhile, I got a 128x64 oled which just works, so I will be shoving this one into recycling can very soon. Feel free to close the issue.

closed 2016-12-13 by rm-hull

sbow, 2018-09-22:
FYI - I was having the same issue with my adafruit ssd1306 128x64 OLED.

After connecting RESET to GND & VIN a few times via a 2.2k resistor, the welcome.py example worked immediately; big change after an hour or so of random pixels.... 

After shutting down the script & restarting it without doing a RESET, it continued to work.

I am have this working now on a NVIDIA Jetson TX1

sbow, 2018-09-22:
testing