rm-hull/luma.oled #71: Create document with devices that work

thijstriemstra, 2016-12-06:
We all have devices that worked and when you're starting out with these displays, having a list of devices that are confirmed to work is a great help. I suggest putting it in the wiki (and make it public to edit) but I'm not sure if it's possible to add images there.

rm-hull, 2016-12-06:
Was considering adding a **benchmarking.rst** to the project, but a public wiki is probably the way to go here, along with a link from the main docs to encourage folks to add their details.

Maybe a tabular format with the following fields:
* Date/time tested
* Known to work or not ( green tick / red cross )
* OLED brand ( Adafruit / Sainsmart / Unbranded / link to purchase page )
* Github user id
* SoC model name & spec ( CPU cores / speed / memory capacity)
* Linux version (uname -a) 
* Python version
* (this) Library version
* Command run
* Resultant FPS
* Notes
* Uploaded image

The actual command run would convey the device class, interface, screen size

e.g. 

```
python examples/perfloop.py -i i2c -d sh1106 --width 96 --height 16
```

```
python examples/perfloop.py -i spi --spi-bus-speed=32000000 -d ssd1306 --width 128 --height 64
```

thijstriemstra, 2016-12-06:
@rm-hull any suggestion for name of wiki page?

rm-hull, 2016-12-06:
Hmmm.. 

https://github.com/rm-hull/ssd1306/wiki/Usage-&-Benchmarking

Tables aren't going to be so easy to maintain - maybe explore alternative layouts or a shared google sheets?

thijstriemstra, 2016-12-06:
I don't think it's a bad start, it can always be improved later, I don't expect > 10 devices to be added anyway. I'll add my results and see where I can improve the page.

thijstriemstra, 2016-12-07:
@rm-hull the entry you added, what dimensions does it have? And does it use the datasheet found in the docs?

rm-hull, 2016-12-07:
dimensions = 128x64, datasheet = yes, i2c bus speed was 400KHz. bounce demo is about 20 FPS

By default the I2C bus speed is 100KHz of RPi, see docs:

> Optionally, to improve performance, increase the I2C baudrate from the default of 100KHz to 400KHz by altering /boot/config.txt to include:
>
> ```
> dtparam=i2c_arm=on,i2c_baudrate=400000
> ```

thijstriemstra, 2016-12-07:
Maybe we should report both, with and without 400khz?

rm-hull, 2016-12-07:
For an original RPi model B (1 core, 700MHz, 512MB) i2c bus speed 400KHz the perfloop comes in at 21.54 FPS and the bounce demo is 9.15 FPS <-- these are all python2, and I'm aware python3 is slower than that

thijstriemstra, 2016-12-07:
For SPI on RPI3 perfloop reports 68.23 FPS and bounce 46.8 (py 3.4.2). I'll add it to the wiki now (aka page is locked till im done ;)

thijstriemstra, 2016-12-07:
Tip: it's possible to clone the wiki: https://github.com/rm-hull/ssd1306.wiki.git (relief, that online editor sucks).

thijstriemstra, 2016-12-07:
It's also impossible to put a link in that table so I'm going to restructure it..

thijstriemstra, 2016-12-07:
@rm-hull the page is unlocked again, my results are added. Will add links to docs in PR now.

commit 05419e8d (2016-12-07): add references to wiki doc (fixes #71)
commit 1ba37c82 (2016-12-08): add references to wiki doc (fixes #71)
closed 2016-12-07 by rm-hull