rm-hull/luma.oled #73: History management

thijstriemstra, 2016-12-06:
Let's say I have a message, an IP address, displayed and an error occurs that needs to be displayed for a few seconds, and then it should revert to the IP address display. I've implemented something hacky myself but we might as well add something to this library. Thoughts?

rm-hull, 2016-12-07:
Yeap, sounds like a good idea.
Something like: 

```python
with canvas(device) as draw:
    draw.text((20, 10), text=myip_addr, fill="white")

...

# An error occurs... make a savepoint which is restored later
device.save()

# Now show the error
with canvas(device) as draw:
    draw.text((20, 10), text="Something bad happened", fill="white")
time.sleep(2)

# Restore the previously saved image, and it auto displays
device.restore()
```

The `save`/`restore` methods would just be a stack, so you could have multiple savepoints

thijstriemstra, 2016-12-08:
How would I get deeper in the stack with multiple savepoints?

rm-hull, 2016-12-08:
```python
with canvas(device) as draw:
    draw.text((20, 10), text="screen1", fill="white")

device.save()
with canvas(device) as draw:
    draw.text((20, 10), text="screen2", fill="white")

device.save()
with canvas(device) as draw:
    draw.text((20, 10), text="screen3", fill="white")

device.save()
with canvas(device) as draw:
    draw.text((20, 10), text="screen4", fill="white")

time.sleep(2)
device.restore()  # shows 'screen3'

time.sleep(2)
device.restore()  # shows 'screen2'

time.sleep(2)
device.restore()  # shows 'screen1'

time.sleep(2)
device.restore()  # stack exhausted - raises an error (preferred) or does nothing

thijstriemstra, 2016-12-08:
device.restore(-2)?

rm-hull, 2016-12-21:
Have pushed a branch *feature/savepoints* which does the above - you wrap an existing device with a `history` class and draw with that (it has the same capabilities as a regular device). Not entirely sure 'history' is the right descriptive word for that class, so may change its name if I can think of anything better. It is up to the caller to manage the calls to `history.restore()` (this can be done by checking `len(history) > 0`, else an index error will result.

There is a sample example, called _savepoint.py_ which has two phases - the first drops every other savepoint (bit like your `device.restore(-2)` idea, but the number is >= 0: it is the number of savepoints to drop before restoring). The second phase reuses the same canvas to draw onto, and restore every point.

[image]

Anyway, see how this works for your code? I'll try and get it into master and a new version before Christmas.

commit 1b4acd7a (2016-12-23): Add savepoint/restore functionality - fixes #73

history class wraps a device, and provides the same capabilities to
draw with, but adds a savepoint() method that pushes the last drawn
image onto a stack, and a restore() method that pops the most recent
image from the stack and redraws it.
commit 80c21523 (2016-12-23): Merge pull request #105 from rm-hull/feature/savepoints

Add savepoint/restore functionality - fixes #73
closed 2016-12-23 by rm-hull