rm-hull/luma.oled #75: Add ability to adjust brightness of screen

rm-hull, 2016-12-07:


thijstriemstra, 2016-12-08:
Is this available for both SPI/I2C? What kind of brightness range are we talking about?

rm-hull, 2016-12-08:
Should be agnostic of SPI/I2C ... page 36 of the SSD1306 spec has this:

> ## 10.1.7 Set Contrast Control for BANK0 (81h)
> This command sets the Contrast Setting of the display. The chip has 256 contrast steps from 00h to FFh. The segment output current increases as the contrast step value increases. 

I'm guessing that the SH1106 has a similar command - if it has - then in the device class, we could add a method `contrast(self, level)` that asserts 0 <= level <= 255 and then calls `self.command(const.SETCONTRAST , level)`

In the `__init__` methods, remove specific command here https://github.com/rm-hull/ssd1306/blob/master/oled/device.py#L222 and same for the sh1106 init, and replace with self.contrast(0xCF) or whatever the sh1106 value shows

Add something to the demo to loop over the contrast values, showing a fade in/out effect

thijstriemstra, 2016-12-08:
> Add something to the demo to loop over the contrast values, showing a fade in/out effect

Reminds me of the plain `demo.py`, the ON/OFF operations should display ON, blank screen, ON, blank screen etc. Now it just blinks a few times, almost looking buggy.

commit 5c04746a (2016-12-11): Add ability to adjust brightness of screen - fixes #75
closed 2016-12-11 by rm-hull