rm-hull/luma.oled #82: Error when PIL gif plugin missing

thijstriemstra, 2016-12-08:
Apparently I have a PIL installed that doesn't have the GIF plugin available. This produces a traceback for the gifanim example on Ubuntu with python 3.5.2:

```
$ python examples/clock.py -d gifanim --max-frames 20
Error in atexit._run_exitfuncs:d GIF
Traceback (most recent call last):
  File "/home/foo/.virtualenvs/bar/lib/python3.5/site-packages/oled/emulator.py", line 133, in write_animation
    append_images=self._images[1:])
  File "/home/foo/.virtualenvs/bar/lib/python3.5/site-packages/PIL/Image.py", line 1683, in save
    format = EXTENSION[ext]
KeyError: ''
```

It should print a warning with explanation instead?

thijstriemstra, 2016-12-08:
It also produced a broken `oled_anim.gif` file of 0 bytes.

rm-hull, 2016-12-08:
Are you on the latest from master?

I saw this the other day, and fixed it in commit https://github.com/rm-hull/ssd1306/commit/3e942e5d84c828aae599bc923e1e1dfbb737ee18

thijstriemstra, 2016-12-08:
Sorry this is an issue with 1.1.0 release only, it's fixed in master, sorry for the noise.

closed 2016-12-08 by thijstriemstra

thijstriemstra, 2016-12-08:
@rm-hull can we put out a new release?

rm-hull, 2016-12-08:
Sure - will do that now - will go for v1.2.0

thijstriemstra, 2016-12-08:
@rm-hull let's bump the classifier from `Development Status :: 4 - Beta` to 5 before or after that release.

thijstriemstra, 2016-12-08:
I suppose it takes a while but pypi is still on 1.1.0 here:

```
Installing collected packages: ssd1306
Successfully installed ssd1306-1.1.0
```

rm-hull, 2016-12-09:
1.2.0 should be there now - hit a bit of a road-bump publishing due to b0036d74a14ac8dc79b6995e0fe99cb93853b2b7

thijstriemstra, 2016-12-09:
Yep. Thx.