rm-hull/luma.oled #85: Discussion about library name and direction

thijstriemstra, 2016-12-08:
Right now this is an OLED centric library called `ssd1306` that also supports SH-something. What about renaming it.. or creating a new library that basically handles *any* RPI compatible display that can render Pillow objects? Afaik max7219 doesn't use pillow, but it's a RPI display, so the Pillow object assumption doesn't hold there, but that's ok, a one-stop display library for Python, that can have dependeices on other RPI display libraries like RPLCD, sounds awesome.

thijstriemstra, 2016-12-08:
If it's renamed the ssd1306 package on pypi is obsolete. I suppose that package could also become some sort of plugin that extends the main/core library.

rm-hull, 2016-12-08:
Was considering renaming this as **RPi.OLED**

* I dont really think it  should be a kitchen sink for all different types of LCDs - supporting the different permutations of SSD1306/SH1106 on different sized displays on different SoC's is difficult enough!
* I have _slight_ reservations about using RPi.OLED only because I dont want it to be exclusively for the Raspberry Pi - it should be able to work on other SoC's, but at the moment we have a hard dependency on RPi.GPIO (for SPI devices) which wont run on anything other than Raspberry Pis
* The vast majority of devices out there _seem_ to be SSD1306, and the github site gets a lot of traffic from google from people searching for "raspberry pi ssd1306" - an argument for keeping it as is, possibly.
* The max7219 driver is a different beast, and you're right it doesn't use Pillow, but it is such low-resolution, I'm not sure how it could take advantage of Pillow effectively, although ... it might work with the viewport stuff though...

thijstriemstra, 2016-12-09:
I'm building a product and it took a lot of time to find all the right libraries, I guess that's my main focus here, having to look for x, y, z libraries (and whether it supports python 3 etc) is a pita.

But yes, RPi.OLED it is!

thijstriemstra, 2017-01-10:
Right now there's this dependency graph:

[image]

And when using max7219 as an example, you mentioned the viewport stuff might work, but having a png emulator would also be nice for max7219. I think the whole emulator concept is really useful for debugging and previewing content on a screen.

rm-hull, 2017-01-10:
Yep, I think you're right. I'm definitely coming round to the idea that supporting max7219 (and pcd8544) would be viable with the pillow display/canvas mechanism and the viewport.

rm-hull, 2017-01-10:
How about extracting all the canvas, viewport and emulator code into a new repo/package called *luma.core*. Then rename this repo/package to *luma.oled* that extends the core package to provide fuctionality for OLED displays, and then add new repos/implementations called *luma.led_matrix* (for max7219 and other discrete-LED displays, eg http://www.ebay.co.uk/itm/351368537705?_trksid=p2055119.m1438.l2649) and *luma.lcd* (for PCD8544 and other small bitmapped LCD displays)

See https://en.wiktionary.org/wiki/luma#Esperanto, meaning 'light' or 'bright', but it fits quite nicely with monochrome roots of the project

thijstriemstra, 2017-01-10:
@rm-hull brilliant!

closed 2017-01-13 by rm-hull