rm-hull/luma.oled #86: Readthedocs.io build failure

rm-hull, 2016-12-09:
Caused by this commit presumably: fe70eaa0a64cde1daa79d7036f029fdad8b607d8 

See https://readthedocs.org/projects/ssd1306/builds/4766214/

```
Running Sphinx v1.3.5
making output directory...

Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/ssd1306/envs/stable/local/lib/python2.7/site-packages/sphinx/cmdline.py", line 243, in main
    opts.warningiserror, opts.tags, opts.verbosity, opts.jobs)
  File "/home/docs/checkouts/readthedocs.org/user_builds/ssd1306/envs/stable/local/lib/python2.7/site-packages/sphinx/application.py", line 144, in __init__
    self.setup_extension(extension)
  File "/home/docs/checkouts/readthedocs.org/user_builds/ssd1306/envs/stable/local/lib/python2.7/site-packages/sphinx/application.py", line 435, in setup_extension
    err)
ExtensionError: Could not import extension sphinx.ext.autosectionlabel (exception: No module named autosectionlabel)

Extension error:
Could not import extension sphinx.ext.autosectionlabel (exception: No module named autosectionlabel)
```

thijstriemstra, 2016-12-09:
Shit. This post told it was safe: http://stackoverflow.com/a/40225340

And it works locally. What's their Sphinx version? Hmm

rm-hull, 2016-12-09:
Fixed

closed 2016-12-09 by rm-hull