rm-hull/luma.oled #90: SSD1322

thijstriemstra, 2016-12-11:
This [3.2 inch OLED](http://www.ebay.com/itm/SPI-3-12-3-2-256x64-Graphic-OLED-Blue-Display-SSD1322-Arduino-PIC-ARM-/161840095968?hash=item25ae6be2e0:g:tf8AAOSwiLdWCBDr) looks awesome, but it's a SSD1322. Can this be supported (easily)?

thijstriemstra, 2016-12-11:
Datasheet: http://www.farnell.com/datasheets/303158.pdf

rm-hull, 2016-12-11:
Looks relatively straightforward. This seems to be a more detailed data sheet though: http://www.newhavendisplay.com/app_notes/SSD1322.pdf

rm-hull, 2016-12-21:
got one of these on order, https://www.aliexpress.com/item/1pcs-3-12-inch-16PIN-Yellow-OLED-Module-SSD1322-Drive-IC-256-64-SPI-8Bit-Parallel/32414142892.html

thijstriemstra, 2016-12-21:
I ordered a white one earlier today!

rm-hull, 2016-12-21:
Note to self: https://en.wikipedia.org/wiki/Grayscale#Luma_coding_in_video_systems

rm-hull, 2017-01-16:
Package arrived today ... standby!

thijstriemstra, 2017-01-16:
Arr! Mine hasn't yet but that's ok.. Would love to see a clip/picture.

thijstriemstra, 2017-01-17:
Mine arrived as well, but there are no pins on it, and soldering looks hard, e.g. have to disassemble it? Thoughts @rm-hull?

rm-hull, 2017-01-17:
Mine looks exactly like the one in https://www.aliexpress.com/item/1pcs-3-12-inch-16PIN-Yellow-OLED-Module-SSD1322-Drive-IC-256-64-SPI-8Bit-Parallel/32414142892.html - 2x8 holes down one side - there are no pins, so it will need some soldering. It doesn't look like it needs to be disassembled though, but must admit I haven't looked in too much detail.  

Are the pins inaccessible, or is the fact you have to do some soldering? If the latter, then try and source some solderless headers like these: https://www.sparkfun.com/products/retired/10527

thijstriemstra, 2017-01-18:
Another picture (with horrible quality, I don't get this phone) that shows the pins are in the center, unlink yours on the side. So this is not a recommended display to buy. Your solderless headers sound like a good idea but it looks like taking it apart temporarily shouldn't be too hard, I'll try that first.

[image]

thijstriemstra, 2017-01-23:
Soldered it, hooked it up, but then realized there's no support yet, but yeah.

rm-hull, 2017-01-23:
This is probably the next task.... I think it should be quite similar to the SSD1325 drivrt

rm-hull, 2017-01-26:
@thijstriemstra Well I've drafted out a driver for this OLED, but am not getting any signs of life on the device at all ... if I push the WIP to a branch, can you try the code with your device?

thijstriemstra, 2017-01-26:
@rm-hull yep sounds good

thijstriemstra, 2017-01-27:
The ebay page also mentions 5V v.s. the 3.3V in our hardware doc.

thijstriemstra, 2017-01-27:
@rm-hull seem results as you (with both 3.3v and 5v): nothing's happening. The ports I used: SCK, MOSI, RES, CS, DC, GND and VCC.

rm-hull, 2017-01-27:
ok, well thats \~sort of\~ encouraging, it is unlikely that both of them are broken, more likely the init sequence is wrong.

Also, on the back side, do you see any jumpers similar to BS1 and BS2 below with R18-R20 0-ohm resistors:

[image]

If so how are they jumpered?

rm-hull, 2017-01-28:
Progres:

[image]

rm-hull, 2017-01-28:
Bingo!

[image]

[image]

rm-hull, 2017-01-28:
```
$ cat ssd1322.conf 
--display=ssd1322
--interface=spi
--width=256
--height=64
--spi-bus-speed=16000000
```

rm-hull, 2017-01-28:
@thijstriemstra:

> My device has the following connections: one the left side there's the 'TF' section with MISO, SCK MOSI, CS and on the right 'SPI' side there are: RES, CS, DC, SCL, SDA, GROUND and VCC.

The 'TF' section probably refers to the unsoldered SD card slot on your device:

[image]

The pic is too fuzzy to make any other details out - you need a better 📷 !!

> Do you have the same connections @rm-hull? 

No, mine is two rows of 8 pins, and is documented like this, and can be configured in SPI or 8-bit parallel mode:

[image]

> I guess I'll use connections from both sides if I look at the connection diagram in our docs.

You should only connect up the pins on the SPI side _only_, wiring exactly like the docs here: https://luma-oled.readthedocs.io/en/latest/hardware.html#spi

It should work with 3V3 or 5V (mine was 3V3)

rm-hull, 2017-01-29:
New demo in [examples repo](https://github.com/rm-hull/luma.examples/blob/master/examples/scrolling_pixelart.py) to showcase colour and greyscale rendering:

[image]

thijstriemstra, 2017-01-29:
Holy shit, wow! Awesome work @rm-hull, as always! Gonna test it now. Thanks for the detailed comments, that'll make good documentation as well.

thijstriemstra, 2017-01-29:
I'm getting this traceback for the examples:

```
Traceback (most recent call last):
  File "/home/pi/projects/pi-test/luma.examples/examples/demo_opts.py", line 77, in <module>
    rotate=args.rotate)
  File "/home/pi/projects/pi-test/luma.oled/luma/oled/device.py", line 284, in __init__
    self.command(0xFD, 0x12)        # Unlock IC
  File "/home/pi/projects/pi-test/luma.oled/luma/oled/device.py", line 347, in command
    self._serial_interface.data(args)
  File "/home/pi/projects/pi-test/luma.core/luma/core/serial.py", line 152, in data
    write(data[i:i + 4096])
TypeError: write() argument 1 must be list, not tuple
```
Hmm

thijstriemstra, 2017-01-29:
And here's a picture:

[image]

rm-hull, 2017-01-29:
Re: traceback ... Haven't tried it on python 3 and there are no tests yet, so not trapped by tox/travis

rm-hull, 2017-01-29:
Fix for python 3 in commit 63814ef

closed 2017-01-29 by rm-hull

thijstriemstra, 2017-01-31:
> You should only connect up the pins on the SPI side only, wiring exactly like the docs here: https://luma-oled.readthedocs.io/en/latest/hardware.html#spi

@rm-hull but my spi side contains SDA and SCL pins and D0 and D1 are not even on there. :(

rm-hull, 2017-01-31:
D0 = SCL (clock)
D1 = SDA (data)

Probably worth adding some wiring info for different devices

rm-hull, 2017-02-02:
@thijstriemstra - did you get your SSD1322 working? I only tried it on a PiZero so far (both Py2.7 and 3.4) and seems to work fine, but it would get glitchy display if I raised the SPI bus speed over 16MHz (this might be something to do with the PiZero... it's not fast)

thijstriemstra, 2017-02-02:
I'll test it later tonight.

thijstriemstra, 2017-02-16:
@rm-hull unfortunately I didn't get it to work and kind of ready to throw it in the bin :-1: raspberrypi is wonderful but the amount of monkeypatching and trickery needed to get 3rdparty stuff working is disheartening.

rm-hull, 2017-02-16:
@thijstriemstra That's a bit disappointing 🙁. It's *possible* that it is configured in parallel mode rather than SPI... I made a comment on this thread about the BS1 and BS2 jumpers -they control the config.

If you want to post it on to me, I'm happy to test it for you, document the wiring connections and then send it back to you.

PR #131 (2017-01-29): Add support for 256x64 4-bit greyscale SSD1322 device
Fixes #90
commit 70346ca9 (2017-01-29): Add support for 256x64 4-bit greyscale SSD1322 device (#131)

Fixes #90 

* Working SSD1322 implementation
* performance improvements
* Update changelog
* Fix for python3
* Add SSD1322 regression-pack tests
* Remove redundant print statement
* Ignore .AppleDouble files
* Add FIXME's for asserts
* Add doc references for SSD1322