rm-hull/luma.oled #112: Examples failing with error: __init__() got an unexpected keyword argument 'rotate'

aaronnewcomb, 2017-01-09:
None of the examples will run out of the box for me. I am using I2C sdd1306 128x64 module on port 1.

./pi_logo.py

usage: pi_logo.py [-h] [--config CONFIG]
                  [--display {ssd1306,ssd1325,ssd1331,sh1106,capture,pygame,gifanim}]
                  [--width WIDTH] [--height HEIGHT] [--rotate {0,1,2,3}]
                  [--interface {i2c,spi}] [--i2c-port I2C_PORT]
                  [--i2c-address I2C_ADDRESS] [--spi-port SPI_PORT]
                  [--spi-device SPI_DEVICE] [--spi-bus-speed SPI_BUS_SPEED]
                  [--bcm-data-command BCM_DATA_COMMAND]
                  [--bcm-reset BCM_RESET]
                  [--transform {none,identity,scale2x,smoothscale}]
                  [--scale SCALE] [--mode {1,RGB,RGBA}] [--duration DURATION]
                  [--loop LOOP] [--max-frames MAX_FRAMES]
pi_logo.py: error: __init__() got an unexpected keyword argument 'rotate'

rm-hull, 2017-01-09:
Hello, did you install from PyPi, and are trying to run the latest examples from the master branch?

If so, could you try:

    $ cd ssd1306
    $ PYTHONPATH=. ./examples/pi_logo.py

[This runs the code directly from the source directory]

If that works, I think the reason is that the published version on PyPi lags slightly behind the master branch - I'll have a think how we can make sure master is always in line with the published version.

thijstriemstra, 2017-01-09:
> I'll have a think how we can make sure master is always in line with the published version.

Use a dev branch and always publish to pypi once it's merged to master?

rm-hull, 2017-01-09:
@thijstriemstra agreed -- I'll release a 1.5.0 branch now to PyPi, and with immediate effect, will add a `develop` branch onto which all PRs should be applied.

rm-hull, 2017-01-09:
@aaronnewcomb - I published 1.5.0 so master & PyPi should be aligned now. If you try:

    $ sudo pip install ssd1306 --upgrade

and try the examples again, let me know how it goes.

aaronnewcomb, 2017-01-10:
Bingo! That did it. I had installed with pip, but then downloaded the master branch from GitHub, so things weren't in sync. BTW - I am including an example of using this great library in my new book - Linux for Makers. Thanks for everyone's hard work on this library.

aaronnewcomb, 2017-01-10:
If I install with pip do I get the examples, too? I couldn't find them anywhere.

closed 2017-01-10 by aaronnewcomb

thijstriemstra, 2017-01-10:
@aaronnewcomb awesome! make sure to mention it needs 1.5.0 or newer in your book ;) And no, using pip does not install the examples..

rm-hull, 2017-01-10:
@aaronnewcomb - glad it worked. Looking forward to seeing the book!