rm-hull/luma.oled #116: Explore possible SSD1306 optimization improvements

rm-hull, 2017-01-13:
Rather than iterating over the buffer that represents the SSD1306's internal memory doing some cranky calculations on the way, iterate over the image data instead, and use some pre-calculated offsets to update the buffer. Doing this on the PCD8544 code-base saw an improvement of 54 FPS to about 102 FPS, see here: https://github.com/rm-hull/luma.lcd/blob/package-rename/luma/lcd/device.py#L73-L80

The code is also somewhat simpler.

Can also be applied to the SH1106 driver as well.

rm-hull, 2017-01-13:
@thijstriemstra I've simplifed the display code for the SSD1306 - I would've assumed that the code would be faster as well (see https://github.com/rm-hull/luma.oled/compare/develop...optimize-display), but it isn't - on my ageing RPi. model B (700MHz 1 core), the I2C version performance dropped from 26FPS to 24FPS. Whereas, I get slightly better performance when running on the Orange Pi Zero.

Would you be able to benchmark the code in the **optimize-display** branch against your 128x64 SPI OLED and document your results here.

I think even at the expense of a few FPS, the refactored code is a bit more readable. 

p.s. The build job has failed because the data is being sent in a different order now (it shows correctly because the segment remapping) - I'll fix that up shortly.

thijstriemstra, 2017-01-14:
I'll try to test it tomorrow @rm-hull. I wonder how a small (optional) C-extension would improve performance, if we can pinpoint the bottleneck.

thijstriemstra, 2017-01-16:
With a RPi 2 and SPI SSD1306 on that branch it reached 48.30 FPS. On master it reached 40.04 FPS so this branch is definitely an improvement @rm-hull.

commit b2d69ff5 (2017-01-16): Simplify SSD1306 display logic - fixes #116 (#124)

* Fix test data - processes the row data in mirror order (the segment remap is reversed)
* Update changelog
closed 2017-01-16 by rm-hull