rm-hull/luma.oled #117: Badge layout in README.rst

rm-hull, 2017-01-13:
@thijstriemstra do you know why the readme badges are not displaying all on one line in github (they used to up until recently):

[image]

The same readme renders properly in PyPi:

[image]

This seems to be something specific to reST documents - markdown readme's in other projects dont seem to be affected

thijstriemstra, 2017-01-15:
Looking at the HTML, it seems that the 'maintained' badge is wrapped in a `p` tag that adds the extra linebreak(s). I'd get rid of the maintained badge or place it as the final badge and accept it will be rendered in it's own `p` tag.

rm-hull, 2017-01-15:
Its something weird to do with rST space or something

closed 2017-01-15 by rm-hull