rm-hull/luma.oled #120: Separate modules for devices

thijstriemstra, 2017-01-14:
This more a nice to have, but in my application I only use one OLED type at any given time. Since I package my app with pyqtdeploy it's easy to exclude python modules, and ideally I would only include my preferred device driver and leave the rest out.

rm-hull, 2017-01-16:
So two observations here:

* I realise we've just made a breaking change by futzing the package namespaces, not sure this justifies another breaking change though
* If we did have separate modules, presumably the would become `import luma.oled.device.ssd1306.ss1306` or slightly less jarring: `from luma.oled.device.ssd1306 import ssd1306`

thijstriemstra, 2017-01-16:
I would keep the import the same, so `from luma.oled.device import ssd1306` and inside `luma.oled.device` something like:

```python
from luma.oled.device.sdd1306 import sdd1306
# etc
```

to preserve backwards-compatibility and keep the imports short but still have the code in separate modules.

Another thing to ponder is the introduction of relative imports but that's a different ticket.

thijstriemstra, 2017-02-02:
are you +1 on this @rm-hull?

rm-hull, 2017-02-02:
What is the motivation for only packaging just the device class you need? - the files aren't huge, and presumably there is negligible runtime impact by including everything? How do you know you've got all the dependencies? For example, would you only include the spi class from core? what happens to your project when we refactor some classes into separate modules that you don't include?

With the scheme you're suggesting, 
* how would the rendered API docs render, especially with the source link? 
* what is the definitive public face of the API (e.g for ssd1306, would it be _luma.oled.device.sdd1306.sdd1306_ or _luma.oled.device.sdd1306_?) 
* would we provide docs that fitted the current interface, or the new interface? 

If there's a clear case for doing it, then it's worth spending some time, but I must admit, I'm struggling to see the benefits.

thijstriemstra, 2017-02-02:
Yeah you're right, it's probably overkill (and it would've made more sense during the first release of this package, it's too late now).

closed 2017-02-02 by thijstriemstra