rm-hull/luma.oled #129: Weird patterns on SSD1306

Doridian, 2017-01-25:
Greetings,

I have just tried out your library on a probably unsupported device (RTk.GPIO, a RPi compatible GPIO header attachable via USB).
It sort of works, but only half the display. And the working half seems a bit, you might say confused.
[image]
Is this something known with "unsupported" devices?

thijstriemstra, 2017-01-25:
@Doridian what command did you use to produce above example? Can you provide a link to the ebay/store page or datasheet document?

rm-hull, 2017-01-25:
Firstly, \~damn\~ I hate it when I see a new gadget, ... i'm now going to have to get one of those RTk.GPIO headers  ;-)

Secondly I dont recognise the device dimensions - are you sure it is a SSD1306? a link would be useful. It _looks_ like it is 256x64...

Thirdly, at least the demo image is rendering, albeit wrong dimensions, glitchy & mirrored, so we'll be able to write a driver for it, most likely it will need a slightly different segment mapping.

Doridian, 2017-01-25:
eBay page is: (Some of it is in German, but nothing that Google Translate can't solve)
http://www.ebay.de/itm/381900747071?_trksid=p2060353.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

The eBay page in the title talks about a TE667 but at the bottom goes on about SSD1306.
Since I could find basically nothing about a TE667, I assume the "SSD1306" would be the correct description.

thijstriemstra, 2017-01-25:
@Doridian and what command/example/script did you use?

Doridian, 2017-01-25:
@thijstriemstra Just edited that into my comment, forgot before:
`demo.py --width 128 --height 32 --interface i2c --display ssd1306 --rotate 0`

thijstriemstra, 2017-01-25:
@Doridian as @rm-hull suggested, try a different width/height?

Doridian, 2017-01-25:
No supported size produces different results (only supported sizes seem to be 128x32, 128x64, 96x16).

It still only renders the top half of the display and leaves the bottom half untouched.
If I change the rotation to 2, things become interesting. It rotates the data previously rendered (glitched/mirrored) on the top half onto the bottom half, then proceeds to only render the top half in the same way previously shown.
That is the only way I managed to affect it at all.

thijstriemstra, 2017-01-25:
What if you bypass that RTk.GPIO header and connect the display directly to the Rpi?

rm-hull, 2017-01-25:
Ok, so it looks like you have the 128x32 device, which we've tested OK in the past.
I assume you installed the latest version with pip? If so, could you try:

    $ sudo pip uninstall luma.oled
    $ sudo pip install luma.oled==2.0.1

This puts on the previous version - I did some 'optimization' in 2.1.0, and this could be a regression.

Doridian, 2017-01-25:
As I suspected, it is a problem of the RTk.GPIO dongle combined with your module.
Directly on an RPi (took a while to dig that back out of my drawers including a working keyboard and screen), it renders all examples without issue.

By the way, to make the luma.core work with the RTk.GPIO you have to replace `import smbus2` with `import RTk.smbus as smbus2`

[image]

rm-hull, 2017-01-26:
Ok, well I guess since it works with a real I2C port, then this points to a buggy implementation in RTk.smbus - it looks like it is dropping bytes somewhere along the way, judging by the glitchy rendering.

closed 2017-01-26 by rm-hull