rm-hull/luma.oled #134: INSTALLATION ERROR DEPENDENCIES NOT MET

perrycannon, 2017-02-05:
I received this error
"The following packages have unmet dependencies:
 libsdl1.2-dev : Depends: libdirectfb-dev (>= 0.9.22) but it is not going to be installed
E: Unable to correct problems, you have held broken packages."

after running the apt-get install
"sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg8-dev libsdl1.2-dev"

I added the missing dependency, received a follow on error for unmet dependencies. Each instance I added the unmet dependencies. The install failed.

Thank you
Perry

rm-hull, 2017-02-05:
Hi Perry, thanks for the bug report - I think that libsdl1.2dev is no longer required (we used to have a dependency on Pygame, but not any more), and Rasbian has moved on. So I updated the docs to reflect that. Let me know if you run into any other issues, but you should just be able to run:

    $ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg8-dev

closed 2017-02-05 by rm-hull

perrycannon, 2017-02-05:
Thank you. I really want to use your library.

Perry

> On Feb 5, 2017, at 10:25 PM, Richard Hull <notifications@github.com> wrote:
> 
> Hi Perry, thanks for the bug report - I think that libsdl1.2dev is no longer required (we used to have a dependency on Pygame, but not any more), and Rasbian has moved on. So I updated the docs to reflect that. Let me know if you run into any other issues, but you should just be able to run:
> 
>    $ sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg8-dev

thijstriemstra, 2017-02-06:
I think we'll eventually run into the same problem with libjpeg8-dev. Wouldn't libjpeg-dev be a safer choice? (and maybe same for libfreetype-dev?)

commit 5b13191d (2017-02-05): Remove  libsdl1.2-dev as a dependency - fixes #134

SDL is only required for specific versions of pygame - since this is no longer a dependency, it can be removed