rm-hull/luma.oled #140: keep image on display after script finished

ibobalo, 2017-03-10:
Orange pi zero, 1.3" SH1101 on i2c0

#### Expected behaviour
i'd like to run python script to show system info (hostname, ip address, etc) once on startup (for example from /etc/rc.local). i'd like the info keep displaying after script done ane python process finished. 

#### Actual behaviour
oled display turns off after script finished. all info disappears.

rm-hull, 2017-03-10:
You need to override the `cleanup()` method in the `sh1106` class. This can be done easily as per the following example:

```python
rom luma.core.serial import i2c
from luma.core.render import canvas

from luma.oled.device import sh1106

def do_nothing(obj):
    pass

serial = i2c(port=0)
device = sh1106(serial)

# override the cleanup method
device.cleanup = do_nothing

hostname = "OrangePi Zero"
ip = "192.11.23.191"

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, fill="black", outline="white")
    draw.text((3, 2), hostname, fill="white")
    draw.text((3, 12), ip, fill="white")
```

rm-hull, 2017-03-10:
I misread sh1101 for sh1106 ... the above override should work for _any_ display driver, so if you are using the ssd1306 driver and it works, carry on using that

ibobalo, 2017-03-10:
thanks a lot. looks like it works as expected

but from my point of view it is not right solution. its not obvious. may be
it would better to add device property cleanupOnExit(bool) default True

rm-hull, 2017-03-10:
Longer term, yes I agree, but there are other implications of not cleaning up properly which need to be balanced across how it works for other devices.
 
Also, I guess I should've qualified that the above was a bit of a crafty hack.

thijstriemstra, 2017-03-29:
Can this be closed?

closed 2017-03-30 by rm-hull