rm-hull/luma.oled #146: SSD1331 selection

sgjava, 2017-04-06:
I noticed the back of my SSD1306 SPI had the marking detailed in http://luma-oled.readthedocs.io/en/latest/hardware.html and it works as expected. This SSD1331 display https://www.amazon.com/DIYmall-Colorful-SSD1331-Resolution-Arduino/dp/B01HHPOD44 doesn't have those marking to determine 4 wire. I believe this is similar to the one I had issues with. How do I determine if a SSD1331 is compatible?

thijstriemstra, 2017-04-06:
I also had an incompatible SSD1331 (or that's what I discovered after it started smoking).

sgjava, 2017-04-06:
Mine actually came up , but single images https://youtu.be/2tAudqj_hak would not stay on the screen (like data was still being written or something. Frame updates flickered https://youtu.be/dfYiWA6Rc8k. Someone on the CHIP forums hacked TheRayTracer and got a display like http://www.ebay.com/itm/0-95-RGB-OLED-LCD-Display-Module-SPI-96X64-Resolution-65K-For-Arduino-SSD1331/131876242334 working. Just wondering if anybody knows of a working SSD1331.

rm-hull, 2017-04-06:
@sgjava - the fact that you got the image displaying as per the youtube vids means that it is wired up properly, and you have a 'compatible' SSD1331 device. I also have an SSD1331 and have it connected to a RPi Zero, and it works perfectly & does not flicker like yours does with the C.H.I.P. - possibly for the sake of $4, you should get hold of a RPi Zero and try and isolate where the problem is - if you can get it working on that OK, then this unequivocally points to the C.H.I.P. being at fault. 
Based on the chat in the other PR you raised (https://github.com/rm-hull/luma.core/pull/71), it would seem to point to a buggy/immature SPI implementation in the C.H.I.P. kernel.

@thijstriemstra - We should probably remove the OLED pin numbering in the hardware page in the docs, and make it clear that the VCC on the display goes to 3V3 on the RPi, GND to GND, etc. The only reason I can think it started smoking is by connecting VCC and GND the wrong way round.

rm-hull, 2017-04-06:
In general for SPI devices, anywhere where you see CLK, SCLK, D0 this usually equates to the clock line, and should be connected to the [SCLK pin](https://pinout.xyz/pinout/pin23_gpio11) on the RPi.

Anywhere where you see D1, SDA, DIN this is the data input line on the device and should be connected to the [MOSI pin](https://pinout.xyz/pinout/pin19_gpio10) (MOSI = Master Out, Slave in) on the RPi.

sgjava, 2017-04-07:
The weird thing is that the SPI version of the SSD1306 works perfectly on the CHIP, thus SPI, power, etc. are all good. I have an ODROID C1 I can use as a counter test if I run into issues this time. CHIP works out of the box (NAND, WIFI bluetooth, etc). Plus I just checked and cannot find a Pi 0 for $10 :)

I noticed on Amazon they said some displays are mislabeled as SSD1331 when they are really SSD1351.  Would this cause and issue (running SSD1351 with SSD1331 driver)?

rm-hull, 2017-04-07:
If they are mislabelled SSD1351 then that could account for it. 

Can you take a couple of close-up in-focus pictures of the front and back of the OLED and attach them to this issue? (you can drag & drop jpegs into the comment field to upload them)

sgjava, 2017-04-07:
I already sent that one back. I'm going to order one from EBay, but they all look like they ship from China. I'll close this out, but keep it as a place holder in case I have an issue with the new SSD1331. I'm getting close and this is the first display I couldn't get working on the CHIP. I have a couple TFT displays to test soon as well. Thanks for your advice.

closed 2017-04-07 by sgjava

rm-hull, 2017-04-07:
Shame that you sent it back... Looking at that amazon listing, one of the pics shows the back of the board:

[image]

And it is listed as a 128x128 SSD1351

rm-hull, 2017-04-07:
This is the one I bought: http://www.ebay.com/itm/401236499230 - have bought loads of stuff from *alice1101983* - never had any problems whatsoever, always quick and accurate descriptions.

thijstriemstra, 2017-04-07:
> @thijstriemstra - We should probably remove the OLED pin numbering in the hardware page in the docs, and make it clear that the VCC on the display goes to 3V3 on the RPi, GND to GND, etc. The only reason I can think it started smoking is by connecting VCC and GND the wrong way round.

Sounds good.

> This is the one I bought: http://www.ebay.com/itm/401236499230 - have bought loads of stuff from alice1101983 - never had any problems whatsoever, always quick and accurate descriptions.

@rm-hull also bought that one and incorrectly hooked up which destroyed it.

sgjava, 2017-04-07:
Crap, the one you bought from EBay looks like the same one I bought on Amazon (and I mean exactly). Maybe it was defective, but now I know we were using the same display. Let me try again and I will do my due diligence testing against my ODROID C1+ using https://github.com/swkim01/RPi.GPIO-OdroidC1 for GPIO if I run into issues.

sgjava, 2017-04-07:
Just bought 2 of these SSD1331 displays on EBay. Now I have to wait a month or so :)

svenyonson, 2017-10-06:
@rm-hull any chance of adding SSD1351 driver support to the luma library so we can use the 128x128 displays?

thijstriemstra, 2017-10-08:
@svenyonson can you send @rm-hull one of these devices to test with?

svenyonson, 2017-10-09:
Hello Thijs & Richard,

This is the display that is mentioned in the thread, it is listed as a SSD1331 but the PCB shows SSD1351. 

I don't have any of these (I'm already happily using the LUMA library want to keep using devices that LUMA supports), but if Richard is willing to put the time in, I'm willing to order one and send it to him - I know you guys put in a lot of your own time, so I'm willing to contribute in this way.

Have a look at this link and if it looks worthwhile, I'll order one for you. 

https://www.aliexpress.com/item/1-5-inch-Colorful-OLED-Module-SSD1331-128x128-Resolution-for-51-STM32-Arduino/32721463778.html?ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10130_10068_10344_10342_10343_10340_10341_10307_10060_10155_10154_10056_10055_10054_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10052_10053_10142_10107_10050_10051_10324_10325_10084_10083_10080_10082_10081_10178_10110_10111_10112_10113_10114_10312_10313_10314_10078_10079_10073-10152,searchweb201603_14,ppcSwitch_2&btsid=cacfb552-7901-4f23-a403-19cf4c168cdb&algo_expid=2865334a-fd69-416c-82b8-05c43e4de04b-0&algo_pvid=2865334a-fd69-416c-82b8-05c43e4de04b <https://www.aliexpress.com/item/1-5-inch-Colorful-OLED-Module-SSD1331-128x128-Resolution-for-51-STM32-Arduino/32721463778.html?ws_ab_test=searchweb0_0,searchweb201602_3_10152_10065_10151_10130_10068_10344_10342_10343_10340_10341_10307_10060_10155_10154_10056_10055_10054_10059_10534_10533_10532_100031_10099_10338_10339_10103_10102_10052_10053_10142_10107_10050_10051_10324_10325_10084_10083_10080_10082_10081_10178_10110_10111_10112_10113_10114_10312_10313_10314_10078_10079_10073-10152,searchweb201603_14,ppcSwitch_2&btsid=cacfb552-7901-4f23-a403-19cf4c168cdb&algo_expid=2865334a-fd69-416c-82b8-05c43e4de04b-0&algo_pvid=2865334a-fd69-416c-82b8-05c43e4de04b>

Thanks,

Steve

rm-hull, 2017-10-09:
Happy to write a driver for the SSD1351, and that aliexpress link looks to be the correct device. You could either send the right amount via https://www.paypal.me/rmhull and I'll order it, or ping me on richard@destructuring-bind.org and I'll let you know the shipping address if you want to order it.

svenyonson, 2017-10-09:
> Happy to write a driver for the SSD1351, and that aliexpress link looks to be the correct device. You could either send the right amount via https://www.paypal.me/rmhull <https://www.paypal.me/rmhull> and I'll order it, or ping me on richard@destructuring-bind.org <mailto:richard@destructuring-bind.org> and I'll let you know the shipping address if you want to order it.

Thanks Richard.

I just paypaled you $19.52. Thanks for your efforts.

Steve

rm-hull, 2017-10-09:
Great stuff - just ordered it.

I will probably open up another issue (much like https://github.com/rm-hull/luma.lcd/issues/47) to keep notes & show progress, etc, whilst dev is in progress. Stuff from aliexpress seems to generally takes 2-3 weeks, so I'll keep you posted when it arrives.

svenyonson, 2017-10-09:
> I will probably open up another issue (much like rm-hull/luma.lcd#47 <https://github.com/rm-hull/luma.lcd/issues/47>) to keep notes & show progress, etc, whilst dev is in progress. Stuff from aliexpress seems to generally takes 2-3 weeks, so I'll keep you posted when it arrives.

Great, thanks Richard.

Cheers,

Steve

rm-hull, 2017-11-01:
@svenyonson - latest version (2.3.0) now supports SSD1351 - don't know if you saw the thread #174 ?

svenyonson, 2017-11-01:
Hi Richard,

I saw the email but didn’t realize that it was finished. Thanks so much for your time, I really appreciate it!

I’ll try to give it a go in the next few weeks - I received my 1351 last week as well.

Cheers,

Steve