rm-hull/luma.oled #147: Sdd support for 64x32 I2C Display

mdmiko, 2017-04-09:
With luma lib, the display works but, y offset seems to be negative

As tested on u8glib, only change from 64x48 is multiplex
from 0x02f to 0x01f

http://www.buydisplay.com/default/0-49-inch-oled-display-module-64x32-pixel-ssd1306-spi-i2c-white-on-black

rm-hull, 2017-04-09:
So if you can confirm changing your local version (e.g `/usr/local/lib/python27/luma/oled/device.py`) of https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L125-L130 to read:

```python
        settings = {
            (128, 64): dict(multiplex=0x3F, displayclockdiv=0x80, compins=0x12),
            (128, 32): dict(multiplex=0x1F, displayclockdiv=0x80, compins=0x02),
            (96, 16): dict(multiplex=0x0F, displayclockdiv=0x60, compins=0x02),
            (64, 48): dict(multiplex=0x2F, displayclockdiv=0x80, compins=0x12),
            (64, 32): dict(multiplex=0x1F, displayclockdiv=0x80, compins=0x12)  # add this line
        }.get((width, height))
```

works, and running (from the **luma.examples** repo):

    $ python examples/demo.py --width=64 --height=32

It should look like (pic would be good):

[image]

Then I will add it support to the library and push out a new release.

mdmiko, 2017-04-09:
> oled/device.py#L125-L130 to read:

thanks, for today I can only run a emulation test: python examples/clock.py
--width 64 --height 32 --display capture
It works: https://transfer.sh/PJF3x/luma_000005.png

thijstriemstra, 2017-04-11:
fixed with #144?

rm-hull, 2017-04-11:
No not fixed. Fix is trivial though: it needs `(64, 32): dict(multiplex=0x1F, displayclockdiv=0x80, compins=0x12)` adding to the init method and then someone to verify that it works on a real device (not an emulator).

mdmiko, 2017-04-11:
It works.

closed 2017-04-12 by rm-hull

rm-hull, 2017-04-12:
@mdmiko - release 2.2.7 of luma.oled should now support 64x32 SSD1306

PR #149 (2017-04-12): Add support for 64x32 SSD1306
Fixes #147