rm-hull/luma.oled #148: ODROID C1+ working

sgjava, 2017-04-11:
I have Luma.OLED working on the ODROID C1+ with the I2C version of the SSD1306! Check out [ODROIDtimus Prime](http://dev.odroid.com/viewtopic.php?f=116&t=26674). I built a sick display post from small nylon nuts, bolts, standoffs and a wire terminal.
[image]

thijstriemstra, 2017-04-11:
Awesome, let's see some youtube clips! ;)

sgjava, 2017-04-11:
I will get something up soon. I'm shooting for the SPI version next. You may see a PR to get the GPIO param in :) I'll have to hack it again for the ODROID GPIO lib.

sgjava, 2017-04-12:
Here's the videos:

https://youtu.be/OvVZJjpFl5g
https://youtu.be/NkwIqndtPjs

rm-hull, 2017-04-12:
@sgjava Great vids, but the font-awesome one looked a little flickery.

Are you using the I2C or SPI version of the SSD1306 display ?

I would be interested to know what the spec of the Odroid C1+ is (CPU speed, cores, RAM, SD card class, etc), and then if you could run the perftest demo, this gives some numbers about the best frame rate

    $ python examples/perfloop.py -f conf/your_settings.conf

sgjava, 2017-04-12:
It's the I2C version at 300K (max setting for port). I'm not seeing flicker in the sense that the display is flickering, only when it transitions to the next character. Do you do page flipping in the code? Part of it could be taking a video of the display like a scan line on a TV? Do these displays operate at 60 Hz?

Specs are here http://www.hardkernel.com/main/products/prdt_info.php?g_code=G143703355573

rm-hull, 2017-04-13:
I think you're probably right, it does look like scan lines not synced with video and display

closed 2017-04-13 by rm-hull

sgjava, 2017-04-13:
Yea, I didn't see the flicker with the naked eye (what's most important). My videos are still way better than most of them out there considering I used mt cell phone.