rm-hull/luma.oled #155: diff_to_previous on ssd1306

combs, 2017-05-07:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

Orange Pi Zero, Armbian

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux dof 3.4.113-sun8i #10 SMP PREEMPT Thu Feb 23 19:55:00 CET 2017 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

Running `perfloop.py` or my applications with ssd1306 over i2c, I am no longer seeing a difference between `framebuffer=diff_to_previous` and `framebuffer=full_frame`

anything special I need to do to enable the `diff_to_previous` behavior?

Or does ssd1306 need the new `luma.core.framebuffer` behavior as seen in ssd1331?
https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L214-L222

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

```
python perfloop.py --display ssd1306 --i2c-port 0 --i2c-address 0x3c --framebuffer diff_to_previous
Testing dislay rendering performance
Press Ctrl-C to abort test

Namespace(backlight_active='low', bgr=False, block_orientation=0, config=None, display='ssd1306', framebuffer='diff_to_previous', gpio=None, gpio_backlight=18, gpio_data_command=24, gpio_reset=25, h_offset=0, height=64, i2c_address='0x3c', i2c_port=0, interface='i2c', mode='RGB', rotate=0, spi_bus_speed=8000000, spi_device=0, spi_port=0, v_offset=0, width=128)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
#### iter =    310: render time = 50.57 ms, frame rate = 19.73 FPS
```

```
python perfloop.py --display ssd1306 --i2c-port 0 --i2c-address 0x3c --framebuffer full_frame
Testing dislay rendering performance
Press Ctrl-C to abort test

Namespace(backlight_active='low', bgr=False, block_orientation=0, config=None, display='ssd1306', framebuffer='full_frame', gpio=None, gpio_backlight=18, gpio_data_command=24, gpio_reset=25, h_offset=0, height=64, i2c_address='0x3c', i2c_port=0, interface='i2c', mode='RGB', rotate=0, spi_bus_speed=8000000, spi_device=0, spi_port=0, v_offset=0, width=128)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64
----------------------------------------
#### iter =    310: render time = 49.75 ms, frame rate = 20.06 FPS
```

rm-hull, 2017-05-07:
The simple explanation is that the ssd1306 implementation does not make use of the different framebuffers, and instead just does a full frame render (and ignores any args it does not care about).

The diff_to_previous framebuffer was an attempt to make devices with color and/or higher resolutions perform better. It has been implemented for:
* ST7735
* SSD1331
* SSD1322

In theory, it _should_ be simple to add the same capabilities to SSD1306, SH1106 and SSD1325... I had implemented the same logic for SSD1325 but the way the memory is mapped in the device made it awkward so I ended up not committing it in the end. I suspect that this might be a similar trapdoor for SSD1306.

Interestingly, on my Orange Pi Zero (same kernel version as yours), I am seeing frame rates of ~20FPS with python3 ... previously (Dec 2016?) I'm pretty certain it was hitting somewhere around 55FPS on python2 on OPi0. Hmmm.... 🤔

combs, 2017-05-07:
thanks!

re: speed... huh! I'm using python2, if it helps any. BTW the two i2c busses are different speeds. i2c-0 is 400khz, i2c-1 is 200khz:
```

root@dof:/sys/bus/platform/devices/twi.0# cat /sys/bus/platform/devices/twi.0/info
pdev->id   = 0 
pdev->name = twi 
pdev->num_resources = 2 
pdev->resource.mem = [0x01c2ac00, 0x01c2afff] 
pdev->resource.irq = 38 
pdev->dev.platform_data.bus_num  = 0 
pdev->dev.platform_data.freqency = 400000 
pdev->dev.platform_data.regulator= 0x  (null) 
pdev->dev.platform_data.regulator_id =  
root@dof:/sys/bus/platform/devices/twi.0# cat /sys/bus/platform/devices/twi.1/info
pdev->id   = 1 
pdev->name = twi 
pdev->num_resources = 2 
pdev->resource.mem = [0x01c2b000, 0x01c2b3ff] 
pdev->resource.irq = 39 
pdev->dev.platform_data.bus_num  = 1 
pdev->dev.platform_data.freqency = 200000 
pdev->dev.platform_data.regulator= 0x  (null) 
pdev->dev.platform_data.regulator_id =  
root@dof:/sys/bus/platform/devices/twi.0# 

```
I'm using the faster bus but seeing that 20fps max.

I might take a crack at implementing the diff_to_previous framebuffer for ssd1306. wish me luck!

thijstriemstra, 2018-02-04:
Any news @combs? Please re-open this (or a new ticket) if you still have issues with this library.

closed 2018-02-04 by thijstriemstra

CRImier, 2018-02-06:
Would it make sense to make a simple diff_to_previous that'd only rewrite the pages? That is, if one of the 8 pages hasn't changed, we don't write that one to the display once again.