rm-hull/luma.oled #156: 2.4 inch oled: which one?

thijstriemstra, 2017-05-09:
I need to buy a larger OLED but I want to make sure I buy something compatible. It seems the ones I found are not supported yet:

- [SSD1309](http://www.ebay.com/itm/2-42-inch-SPI-I2C-OLED-LCD-Screen-12864-OLED-display-module-SSD1309-Blue-/181906236021)
- [SSD1305](http://www.ebay.com/itm/Green-2-4-inch-128x64-OLED-Display-Module-I2C-Serial-SPI-SSD1305-w-Tutorial-/291844721833)

Which one would be the easiest to implement?

rm-hull, 2017-05-10:
I would be inclined to get the first one (SSD1309) as this has the breakout board, rather than just being a raw display.

If you want a cheaper alternative (and dont mind LCD) have a look at https://github.com/rm-hull/luma.lcd/issues/46 and branch https://github.com/rm-hull/luma.lcd/tree/st7920 (which is currently WIP).

Alternatively, there is _already_ support for this 2.7" OLED: https://learn.adafruit.com/2-7-monochrome-128x64-oled-display-module/overview (SSD1325)

thijstriemstra, 2017-05-10:
Thanks for the feedback. The watching angle is too important with anything else than oled in my experience. That 2.7 oled does not have enough height unfortunately.

closed 2017-05-14 by thijstriemstra