rm-hull/luma.oled #158: SSD1306 instant message then nothing on Orange Pi Zero...

scargill, 2017-06-03:
Hi

Looking for ideas - I've installed this along with WiringOP on an Orange Pi Zero running DietPi   (formerly had it running on a NanoPI without issue).

I works - kind of...

If I take the SSD1306 example - I have to switch to port 0 as I did on the NEO - no problem... I2c detect shows the device is on the right address.... The device has 2k2 pullups which have worked elsewhere.

So - when I try the simple example that draws a rectangle then hello world, it KIND of works.... the rectangle draws - and the text draws but both then instantly disappear.  Indeed rectangle sometimes does not get finished (but no stray dots or lines) and the few times I can see the text it is perfectly formed... but it is as if there is a clear screen immediately after the drawing (which there is not).

Any ideas?

scargill, 2017-06-03:
This might help...

I put it in a file....
ssd1306.py

If I run that - I get the issue of the text disappearing.. if I put a 5 seconds delay on the end - it works perfectly until 5 second later when it clears ......???

rm-hull, 2017-06-03:
> Looking for ideas - I've installed this along with WiringOP on an Orange Pi Zero running DietPi (formerly had it running on a NanoPI without issue).
>
> I works - kind of...
>
> If I take the SSD1306 [...]

> This might help...
>
> I put it in a file....
> ssd1306.py
> 
> If I run that - I get the issue of the text disappearing.. if I put a 5 seconds delay on the end - it works perfectly until 5 second later when it clears ......???

This behaviour is by design: when the program ends, it runs a cleanup job that puts the OLED display in a low-power state (which effectively turns the screen off) and closes any open devices/channels.

If you don't want this happening either (1) keep the program running, or (2) have a look at issue #140, and also: https://github.com/rm-hull/luma.core/issues/74.

closed 2017-06-03 by rm-hull

scargill, 2017-06-03:
I did thanks – and solved the problem…. Thanks.

 

If you don't want this happening either (1) keep the program running, or (2) have a look at issue #140 <https://github.com/rm-hull/luma.oled/issues/140> , and also: rm-hull/luma.core#74 <https://github.com/rm-hull/luma.core/issues/74> .