rm-hull/luma.oled #159: 32 line display issue

scargill, 2017-06-08:
I'm sure this is an old one..  I have the library running  SSD1306 displays - no problem in Python on a NEO AIR - works a treat.  I put a bounding rectangle at 128*32 and some text in there... waiting for my 128*32 display to arrive. 
When I plugged in the display, the top bar is missing. thee bottom bar seems to be in the right place but the text is squashed up at the top as if every other pixel line is missing....

Ideas?  I really need the 32 line display to work - as the larger 64 line display won't fit in a box I'm using.

rm-hull, 2017-06-08:
* Is it definitely an SS1306 rather than an SH1106 ?
* Can you share the code you are using to draw the rectangle ?
* Can you run the standard demo from the [luma.examples](https://github.com/rm-hull/luma.examples) repo and upload a pic of the result. Use the following (you might need to add other flags if you are using SPI or the I2C port is different on the NEO AIR):

```bash
$ python examples/demo.py --device=ssd1306 --width=128 --height=32
```

I haven't got one of the 128x32 devices to test with (but @thijstriemstra has) - but we did the fix in #23 as far as i remember. Also PR #153 might be relevant if you are using an SH1106.

rm-hull, 2017-06-08:
I take it back, ... I have got a 128x32 ssd1306 somewhere - courtesy of @thijstriemstra .

scargill, 2017-06-08:
HELLO - sorry - email been dead - found a solution...

Here is my start code and why 

serial = i2c(port=0, address=0x3C) # that is fairly standard - some of my boards needed port 1
device = ssd1306(serial, height=32) # if you add that height=32 - MAGICALLY the 32-line board works
device.cleanup = do_nothing  # this is to stop the screen blanking 

The more I use this the more I like it.

rm-hull, 2017-06-08:
Yes, you need to set the height: it will default to 128x64 otherwise.
For consistency, you should probably call: `device = ssd1306(serial, width128, height=32)` just so it is super clear.

> The more I use this the more I like it.
Awesome! the main purpose is to have fun :-)

closed 2017-06-08 by rm-hull