rm-hull/luma.oled #161: can not manage Accented Characters

benalb, 2017-07-04:
#### Type of Raspberry Pi
orange pi zero

#### Linux Kernel version
Linux orangepizero 3.4.113-sun8i #18 SMP PREEMPT Thu Jun 15 02:16:06 CEST 2017 armv7l GNU/Linux

   ...: from luma.core.interface.serial import i2c
   ...: from luma.oled.device import ssd1306 
   ...: from PIL import ImageFont
   ...: from luma.core.virtual import terminal
   ...: 
   ...: serial = i2c(port=1, address=0x3C)
   ...: device = ssd1306(serial)
   ...: font = ImageFont.truetype('/home/foobar/fonts/Ubuntu-C.ttf', 16)
   ...: term = terminal(device, font)

In [3]: term.println('á')
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-3-d3613da199c0> in <module>()
----> 1 term.println('á')

/usr/local/lib/python2.7/dist-packages/luma/core/virtual.pyc in println(self, text)
    265                 self.newline()
    266         else:
--> 267             self.puts(text)
    268             self.newline()
    269 

/usr/local/lib/python2.7/dist-packages/luma/core/virtual.pyc in puts(self, text)
    280         """
    281         for method, args in ansi_color.find_directives(text, self):
--> 282             method(*args)
    283 
    284     def putch(self, char):

/usr/local/lib/python2.7/dist-packages/luma/core/virtual.pyc in putch(self, char)
    301 
    302         else:
--> 303             assert(32 <= ord(char) <= 127)
    304 
    305             w = self.font.getsize(char)[0]

AssertionError:

rm-hull, 2017-07-04:
I suppose if the font being used supports extended character sets, then there is no real need for that assert to be so limiting. 

Could you try removing line 303 from your installation's */usr/local/lib/python2.7/dist-packages/luma/core/virtual.py* and see if that works for you?

In the meantime I will write a test for this and check that it works as to be expected and should be able to push a fix out for it fairly quickly.

Thanks for taking the time to report the issue :+1:

benalb, 2017-07-04:
Thank you for your answer.

Comenting out the 303 line in virtual.py makes the error gone, but the accented character  are displayed strangely: 
á shows in the display as ~Ai  and so..

rm-hull, 2017-07-04:
I notice you're using python 2, and the test case I constructed shows the same  as you.
Python 3 solves a lot of these unicode issues out-of-the-box, and running the test case under python 3 works as expected - shows á rather than as ~Ai.

I will continue to look to see if there is a way we can easily support both python 2 and 3

benalb, 2017-07-04:
Yeah, I've installed python3 version and works ok. 
Again, thank you very much.

thijstriemstra, 2017-07-04:
Ok. So we want to support py2 on this or.. id say dont bother and close this.

rm-hull, 2017-07-04:
Still need to remove the assert and even on py3 there is still a regex bug to do with the ANSI colour codes and unicode. See commit a6a7cc2964e394aaf028ae7c927530e2b56fed38

rm-hull, 2017-07-06:
luma.core 0.9.5 has been released onto pypi (upgrade with `sudo -H pip install -U luma.core`) - this removes the assert, but doesn't fix the regex bug or make it work on python 2

thijstriemstra, 2018-03-22:
> luma.core 0.9.5 has been released onto pypi (upgrade with sudo -H pip install -U luma.core) - this removes the assert, but doesn't fix the regex bug or make it work on python 2

This should be fixed in luma.core > 1.7.0.