rm-hull/luma.oled #162: Hi there

scargill, 2017-07-25:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

scargill, 2017-07-25:
Hi there - had troubles with some non-raspberry Pi boards with LUMA so found your sd1306 code instead. Works a treat... problem is - the DEMOS you point to - are the LUMA demos and they use different includes - so for example I can't find the include to get the TERMINAL demo to run ???

rm-hull, 2017-07-25:
Hi Peter, the luma code has evolved out of the old ssd1306 package (which is now no longer maintained). The terminal and a lot of the other more-recent functionality is specific to the luma codebase only.

If you have an SSD1306 screen, the luma codebase _should_ work (and has been tested) on a number of different SBCs - if you let me know the specific problems you are encountering: using the luma libraries is the way forward.

scargill, 2017-07-25:
Hi there

 

The Luma – last line in the setup – causes problems on for example the FriendlyArm NEO2 – I tried several times – on Ubuntu (up to date) – and when I did get it working – subsequent apt-get upgrades caused a “segmentation error” – this is the only time I’ve seen this and it is not happening with the ssd1306 version – sure I’d love to use the luma version instead… but time after time I’ve met with issues I can’t attribute to anything but… 

 

So here https://tech.scargill.net/ssd1306-on-the-neo-plus2/   I’ve blogged what I’ve done – the one line I changed from the original luma setup is the last one – read here

 

https://tech.scargill.net/ssd1306-on-the-neo-plus2/

 

With the SSD 1306 – no problems loading. 

Happy to hear what I may be doing wrong…  when I tried the LUMA line on a NEO PLUS2 (H5) board I ended up with a big paragraph in bright red ..

 

Happy to be shown to be wrong – but the SSD version works on NEO AIR, NEO2 and NEO PLUS2 using Ubuntu.

 

Pete.

scargill, 2017-07-25:
I don’t know if it is relevant – but the NEO boards use FriendlyArm’s implementation of WIRINGPI called WIRINGNP – something I pushed them hard to implement – I don’t know if that could have any bearing on the problems I had with the LUMA library.

rm-hull, 2017-07-25:
Can you paste a transcript of what was output when the segmentation error occurred?

Also what linux distro are you using ? the ubuntu core or debian?

scargill, 2017-07-25:
Hi

 

Ubuntu Core -  Linux 4.11.2 I believe -  

 

Can’t do a transcript tonight – but I have another NEO2 that needs setting up – assuming I get time tomorrow – I’ll set that up and send you the output…..

 

Thanks

 

Pete.

scargill, 2017-07-26:
Ok so first line...
sudo apt-get install python-dev python-pip libfreetype6-dev libjpeg-dev
no problem

Second line... 
sudo -H pip install --upgrade pip
seems ok

Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Not uninstalling pip at /usr/lib/python2.7/dist-packages, outside environment /usr
Successfully installed pip-9.0.1

Third line...
sudo apt-get purge python-pip
 seems ok

The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-pip-whl
  python-pkg-resources python-setuptools python-wheel
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  python-pip*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 635 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 72733 files and directories currently installed.)
Removing python-pip (8.1.1-2ubuntu0.4) ...

At this point I could install the SSD1306 code without issue but if I try luma...
sudo -H pip install --upgrade luma.oled

(it worries me that it seems to be downloading RPi-GPIO - as this is not for an RPI and I already have FriendlyArm's GPIO utility ????)

~Big red warnings...

Collecting luma.oled
  Downloading luma.oled-2.2.10-py2.py3-none-any.whl
Collecting luma.core>=0.9.0 (from luma.oled)
  Downloading luma.core-0.9.5-py2.py3-none-any.whl (49kB)
    100% |████████████████████████████████| 51kB 613kB/s
Collecting monotonic (from luma.core>=0.9.0->luma.oled)
  Downloading monotonic-1.3-py2.py3-none-any.whl
Collecting RPi.GPIO (from luma.core>=0.9.0->luma.oled)
  Downloading RPi.GPIO-0.6.3.tar.gz
Collecting smbus2 (from luma.core>=0.9.0->luma.oled)
  Downloading smbus2-0.1.5.tar.gz
Collecting pillow>=4.0.0 (from luma.core>=0.9.0->luma.oled)
  Downloading Pillow-4.2.1.tar.gz (12.7MB)
    99% |████████████████████████████████| 12.7MB 1.2MB/s eta 0:00:01Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/dist-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/usr/local/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/local/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/local/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 882, in _download_http_url
    _download_url(resp, link, content_file, hashes)
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 603, in _download_url
    hashes.check_against_chunks(downloaded_chunks)
  File "/usr/local/lib/python2.7/dist-packages/pip/utils/hashes.py", line 46, in check_against_chunks
    for chunk in chunks:
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 571, in written_chunks
    for chunk in chunks:
  File "/usr/local/lib/python2.7/dist-packages/pip/utils/ui.py", line 139, in iter
    for x in it:
  File "/usr/local/lib/python2.7/dist-packages/pip/download.py", line 560, in resp_read
    decode_content=False):
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/response.py", line 357, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/response.py", line 314, in read
    data = self._fp.read(amt)
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/cachecontrol/filewrapper.py", line 63, in read
    self._close()
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/cachecontrol/filewrapper.py", line 50, in _close
    self.__callback(self.__buf.getvalue())
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/cachecontrol/controller.py", line 275, in cache_response
    self.serializer.dumps(request, response, body=body),
  File "/usr/local/lib/python2.7/dist-packages/pip/_vendor/cachecontrol/serialize.py", line 87, in dumps
    ).encode("utf8"),
MemoryError
pi@neo22:~$

scargill, 2017-07-26:
On NEO Plus2 with 1GB Ram similar red warnings (and not working) - but I don't recall the memory error issue...  the NEO I think has 512MB.

rm-hull, 2017-07-26:
Try 

    sudo -H apt --no-cache-dir install luma.oled

How much free memory do you have?
It is probably baulking trying to build and install pillow. All the other dependencies are small python libs.

If that doesn't work, try to install pillow first:

    sudo -H apt --no-cache-dir install pillow
    sudo -H apt --no-cache-dir install luma.oled

scargill, 2017-07-26:
Hi there

 

The NEO PLUS2 has exactly the same issue – and it has 1 GIG of RAM – just like the Pi.

And your SD1306 code which does work – also I believe install Pillow…

 

However looking at your instructions below… 

 

pi@neo22:~$ sudo -H apt --no-cache-dir install pillow

E: Command line option --no-cache-dir is not understood in combination with the other options

 

 

pi@neo22:~$ sudo -H apt install pillow

Reading package lists... Done

Building dependency tree

Reading state information... Done

E: Unable to locate package pillow

pi@neo22:~$

 

pi@neo22:~$ sudo -H apt install luma.oled

Reading package lists... Done

Building dependency tree

Reading state information... Done

E: Unable to locate package luma.oled

E: Couldn't find any package by glob 'luma.oled'

E: Couldn't find any package by regex 'luma.oled'

pi@neo22:~$

scargill, 2017-07-26:
Maybe you meant pip install – well for Pillow that had errors – but with the no-cache dir – still errors – but..

 

 

pi@neo22:~$ pip --no-cache-dir install pillow

Collecting pillow

  Downloading Pillow-4.2.1.tar.gz (12.7MB)

    100% |████████████████████████████████| 12.7MB 1.4MB/s

Collecting olefile (from pillow)

  Downloading olefile-0.44.zip (74kB)

    100% |████████████████████████████████| 81kB 1.8MB/s

Installing collected packages: olefile, pillow

  Running setup.py install for olefile ... error

    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-HUraTQ/olefile/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yOhD9e-record/install-record.txt --single-version-externally-managed --compile:

    running install

    running build

    running build_py

    creating build

    creating build/lib.linux-aarch64-2.7

    copying OleFileIO_PL.py -> build/lib.linux-aarch64-2.7

    creating build/lib.linux-aarch64-2.7/olefile

    copying olefile/olefile.py -> build/lib.linux-aarch64-2.7/olefile

    copying olefile/__init__.py -> build/lib.linux-aarch64-2.7/olefile

    copying olefile/README.rst -> build/lib.linux-aarch64-2.7/olefile

    copying olefile/README.html -> build/lib.linux-aarch64-2.7/olefile

    copying olefile/LICENSE.txt -> build/lib.linux-aarch64-2.7/olefile

    copying olefile/CONTRIBUTORS.txt -> build/lib.linux-aarch64-2.7/olefile

    running install_lib

    copying build/lib.linux-aarch64-2.7/OleFileIO_PL.py -> /usr/local/lib/python2.7/dist-packages

    error: [Errno 13] Permission denied: '/usr/local/lib/python2.7/dist-packages/OleFileIO_PL.py'

 

    ----------------------------------------

Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-HUraTQ/olefile/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-yOhD9e-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-HUraTQ/olefile/

 

 

EVENTUALLY I got it going with 

 

sudo -H pip install --no-cache-dir --upgrade luma.oled

 

then when installing the examples – had to install setuptools as that is apparently not installed.

 

Question….

 

sudo python /home/pi/luma.examples/examples/test.py  --i2c-port 0

 

It was easy in the old code – in the new code – how do I get that PORT setting in my program instead of on the command line? – this no longer seems to have any effect.

 

serial = i2c(port=0, address=0x3C)

 

 

So in my original code using SSD1306 I could do…

 

serial = i2c(port=0, address=0x3C)

device = ssd1306(serial, height=32)

 

now it is…

 

device = get_device()

 

 

how would you add the height into the new versio and the port into the serial definition (tried above – does not work).

rm-hull, 2017-07-26:
Yes, sorry, I did mean _pip_ (not _apt_), so apologies for that.
It sounds like the python install for the neo linux distro is a bit botched TBH.

The `get_device()` method is just a helper method for the demos.
If you look at the API docs (https://luma-oled.readthedocs.io/en/latest/api-documentation.html), you can see that it is possible to override the `width=` and `height=`, so you should be able to still initialize with the following:

```python
serial = i2c(port=0, address=0x3C)
device = ssd1306(serial, height=32)
```

scargill, 2017-07-26:
Ok, I’ll give that a go – one step forward… thanks…

scargill, 2017-08-04:
Hi

 

The reason I stopped installing Luma in the first place – was that ultimately it would cause a segmentation error which prevented me from upgrading.  I thought it was just the NEOs but it’s happening on the Orange Pi Zero.

 

So I upgraded my systems to the Luma code and left them along to run. At the time there were no upgrades so I didn’t see a problem but on the Orange Pi Zero, I’ve just now noted it wanting upgrades….. and it can’t.

 

See the code here – part of a standard apt-get upgrade…

 

 

       libdbusmenu-glib.so.4 -> libdbusmenu-glib.so.4.0.12

        liblz4.so.1 -> liblz4.so.1.7.1

        libwbclient.so.0 -> libwbclient.so.0.12

        libindicator3.so.7 -> libindicator3.so.7.0.0

        libgssapi.so.3 -> libgssapi.so.3.0.0

        libpanel.so.5 -> libpanel.so.5.9

        libgd.so.3 -> libgd.so.3.0.0

        libnl-route-3.so.200 -> libnl-route-3.so.200.22.0

        libgnutls-openssl.so.27 -> libgnutls-openssl.so.27.0.2

        librsvg-2.so.2 -> librsvg-2.so.2.40.13

        libndr-krb5pac.so.0 -> libndr-krb5pac.so.0.0.1

        libvpx.so.3 -> libvpx.so.3.0.0

        libhx509.so.5 -> libhx509.so.5.0.0

        libasan.so.2 -> libasan.so.2.0.0

        libfontenc.so.1 -> libfontenc.so.1.0.0

        libsvn_fs_x-1.so.1 -> libsvn_fs_x-1.so.1.0.0

        librtmp.so.1 -> librtmp.so.1

        libatomic.so.1 -> libatomic.so.1.1.0

        libatk-1.0.so.0 -> libatk-1.0.so.0.21809.1

        libcups.so.2 -> libcups.so.2

        libsvn_diff-1.so.1 -> libsvn_diff-1.so.1.0.0

        libtxc_dxtn_s2tc.so.0 -> libtxc_dxtn_s2tc.so.0.0.0

        libfftw3_omp.so.3 -> libfftw3_omp.so.3.4.4

        libicutu.so.55 -> libicutu.so.55.1

        libxcb.so.1 -> libxcb.so.1.1.0

        libXi.so.6 -> libXi.so.6.1.0

[... 161 lines cut ...]
 

 

So in my original code using SSD1306 I could do…

 

serial = i2c(port=0, address=0x3C)

device = ssd1306(serial, height=32)

 

now it is…

 

device = get_device()

 

 

how would you add the height into the new versio and the port into the serial definition (tried above – does not work).

rm-hull, 2017-08-05:
Can't really see why it's failing from the logs you posted. There's no specific reason why installing Lima.oled should cause install problems or segmentation faults - it is just a pure python library: there are no low level shared library changes.

It is more likely that you are running out of memory, and this is manifesting as baroque segmentation faults. When you get these errors, run `dmesg -T` and see if there are any kernel messages at the time you are installing.

The orange pi zero's are tantalisingly cheap (I have 3) but I have found they are woefully slow and the armbian distro is not a patch on raspian

rm-hull, 2017-08-05:
I've just installed luma.oled on a fresh OPi.Zero machine running the armbian debian jessie image, and it went through without any issues. 

Compiling pillow used all four cores and a lot of memory, but it did work in the end.

```
 $ uname -a
Linux orangepizero 3.4.113-sun8i #18 SMP PREEMPT Thu Jun 15 02:16:06 CEST 2017 armv7l GNU/Linux
```

scargill, 2017-08-05:
Did for me - everything works - then sometime later when you come to do an apt-get upgrade - and there is something to upgrade - always the same story (Opi Zero and NEO2 so far) - eventually - the segmentation error and no way to do an upgrade.   I'd love to hear back in a few days once an upgrade is needed..

scargill, 2017-08-05:
The most humble apologies possible - I've just realised it is not the Orange Pi Zero it is doing it on - it is the NEO PLUS 2 and I also got the same issue on a pair of NEO 2 machines.... I tried reducing resources when doing the upgrade - noting earlier comments - by stopping Node-Red - made no difference.

      libXi.so.6 -> libXi.so.6.1.0
        libdcerpc-server.so.0 -> libdcerpc-server.so.0.0.1
        libsvn_fs_fs-1.so.1 -> libsvn_fs_fs-1.so.1.0.0
        libasn1.so.8 -> libasn1.so.8.0.0
        liblua5.1-c++.so.0 -> liblua5.1-c++.so.0.0.0
        libjpeg.so.8 -> libjpeg.so.8.0.2
        libpangoft2-1.0.so.0 -> libpangoft2-1.0.so.0.3800.1
        libgnutls.so.30 -> libgnutls.so.30.6.2
/usr/lib/aarch64-linux-gnu/mesa-egl:
        libEGL.so.1 -> libEGL.so.1.0.0
/usr/lib/aarch64-linux-gnu/libfakeroot:
        libfakeroot-0.so -> libfakeroot-tcp.so
/usr/local/lib:
        libwiringPiDev.so -> libwiringPiDev.so.2.0
        libwiringPi.so -> libwiringPi.so.2.0
/lib:
Segmentation fault
dpkg: error processing package libc-bin (--configure):
 subprocess installed post-installation script returned error exit status 139
Errors were encountered while processing:
 libc-bin
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@plus22:/$

scargill, 2017-08-05:
Earlier on in the log - this.. no idea if it is relevant.  I NEVER get this on these machines until I start messing with Luma to support SSD1306 display (I like the idea of using one as a general monitor).

Do you want to continue? [Y/n] y
Preconfiguring packages ...
Setting up libc-bin (2.23-0ubuntu9) ...
Segmentation fault
/sbin/ldconfig.real: Path `/lib/aarch64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib/aarch64-linux-gnu' given more than once
/lib/aarch64-linux-gnu:
        libdevmapper.so.1.02.1 -> libdevmapper.so.1.02.1

rm-hull, 2017-08-05:
If you run `dmesg -T` at the bash prompt on your NEO2 directly after experiencing the segmentation fault, do you see anything pertinent?

scargill, 2017-08-05:
Wow – certainly do – lots of it in red as well..

I got lots to do with I2c but thought as I’d been messing with i2c I’d better ensure that had nothing to do with it – so I rebooted the board (user pi).

 

I did the dmesg –T command…

 

pi@plus22:~$ dmesg -T

[Sat Aug  5 22:30:53 2017] Booting Linux on physical CPU 0x0

[Sat Aug  5 22:30:53 2017] Linux version 4.11.2 (root@wwd) (gcc version 6.3.1 20                                                             170109 (Linaro GCC 6.3-2017.02) ) #267 SMP Fri Jun 30 12:03:20 CST 2017

[Sat Aug  5 22:30:53 2017] Boot CPU: AArch64 Processor [410fd034]

[Sat Aug  5 22:30:53 2017] Machine model: FriendlyElec NanoPi-NEO-Plus2

[Sat Aug  5 22:30:53 2017] cma: Reserved 16 MiB at 0x000000007f000000

[Sat Aug  5 22:30:53 2017] On node 0 totalpages: 262144

[Sat Aug  5 22:30:53 2017]   DMA zone: 4096 pages used for memmap

[Sat Aug  5 22:30:53 2017]   DMA zone: 0 pages reserved

[Sat Aug  5 22:30:53 2017]   DMA zone: 262144 pages, LIFO batch:31

[Sat Aug  5 22:30:53 2017] psci: probing for conduit method from DT.

[Sat Aug  5 22:30:53 2017] psci: PSCIv0.2 detected in firmware.

[Sat Aug  5 22:30:53 2017] psci: Using standard PSCI v0.2 function IDs

[Sat Aug  5 22:30:53 2017] psci: Trusted OS migration not required

[Sat Aug  5 22:30:53 2017] percpu: Embedded 23 pages/cpu @ffff80003ef77000 s5621                                                             6 r8192 d29800 u94208

[Sat Aug  5 22:30:53 2017] pcpu-alloc: s56216 r8192 d29800 u94208 alloc=23*4096

[Sat Aug  5 22:30:53 2017] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3

[Sat Aug  5 22:30:53 2017] Detected VIPT I-cache on CPU0

[Sat Aug  5 22:30:53 2017] CPU features: enabling workaround for ARM erratum 845                                                             719

[Sat Aug  5 22:30:53 2017] Built 1 zonelists in Zone order, mobility grouping on                                                             .  Total pages: 258048

[Sat Aug  5 22:30:53 2017] Kernel command line: earlyprintk root=/dev/sda1 rootf                                                             stype=ext4 rw rootwait fsck.repair=yes panic=10 fbcon=map:0

[Sat Aug  5 22:30:53 2017] log_buf_len individual max cpu contribution: 4096 byt                                                             es

[Sat Aug  5 22:30:53 2017] log_buf_len total cpu_extra contributions: 12288 byte                                                             s

[Sat Aug  5 22:30:53 2017] log_buf_len min size: 16384 bytes

[Sat Aug  5 22:30:53 2017] log_buf_len: 32768 bytes

[Sat Aug  5 22:30:53 2017] early log buf free: 14840(90%)

[Sat Aug  5 22:30:53 2017] PID hash table entries: 4096 (order: 3, 32768 bytes)

[Sat Aug  5 22:30:53 2017] Dentry cache hash table entries: 131072 (order: 8, 10                                                             48576 bytes)

[Sat Aug  5 22:30:53 2017] Inode-cache hash table entries: 65536 (order: 7, 5242                                                             88 bytes)

[Sat Aug  5 22:30:53 2017] Memory: 995992K/1048576K available (8254K kernel code                                                             , 590K rwdata, 2768K rodata, 448K init, 393K bss, 36200K reserved, 16384K cma-re                                                             served)

[... 1394 lines cut ...]
[Sat Aug  5 22:34:44 2017] PC is at 0x404c1c

[Sat Aug  5 22:34:44 2017] LR is at 0x404b90

[Sat Aug  5 22:34:44 2017] pc : [<0000000000404c1c>] lr : [<0000000000404b90>] pstate: 60000000

[Sat Aug  5 22:34:44 2017] sp : 0000ffffe9ad84c0

[Sat Aug  5 22:34:44 2017] x29: 0000ffffe9ad84c0 x28: 0000ffffe9ad8630

[Sat Aug  5 22:34:44 2017] x27: 0000ffff8d115000 x26: 0000000000000000

[Sat Aug  5 22:34:44 2017] x25: 0000000026eb8c03 x24: 0000ffff8d115000

[Sat Aug  5 22:34:44 2017] x23: 0028000700352534 x22: 0000ffffe9ad96e8

[Sat Aug  5 22:34:44 2017] x21: 002900068d315034 x20: 0000000000000043

[Sat Aug  5 22:34:44 2017] x19: 0000ffffe9ad96e0 x18: 00000000004a3000

[Sat Aug  5 22:34:44 2017] x17: 000000000000000b x16: 0000000000000000

[Sat Aug  5 22:34:44 2017] x15: 0000000000000010 x14: 0000000000000000

[Sat Aug  5

scargill, 2017-08-24:
Hi Richard

 

I’m 99% sure the common demoninator for these SEGMENT failts is Luma…  something about SO files… and it seems to be boards that have their own version of WiringPi  - I’m going WAY off on a limb here – but does the Luma installation add WiringPi by any chance? I’m wondering if something goes wrong with one of the SO files…

 

I say that as I’ve just installed a NEO PLUS22 now and all works well, Luma works a treat but the first time I came to do an apt-get upgrade…

 

Relevant extracts..

 

Lots and lots of updates and then..

 

        libfakeroot-0.so -> libfakeroot-tcp.so

/usr/local/lib:

        libwiringPiDev.so -> libwiringPiDev.so.2.0

        libwiringPi.so -> libwiringPi.so.2.0

/lib:

Segmentation fault

dpkg: error processing package libc-bin (--configure):

subprocess installed post-installation script returned error exit status 139

 

It always involves wiringPi somehow…

 

Pete.

rm-hull, 2017-08-25:
No, it does not use wiringpi. You can see all the dependencies by looking at the setup.py.
Something else must be installing wiringpi on your neo.

thijstriemstra, 2017-08-26:
What an enormous thread.

Heltec-Aaron-Lee, 2017-12-14:
the purge python-pip made some worng
before `sudo -H pip install –upgrade luma.oled`
you need install again:
`sudo apt-get install python-pip`

http://www.heltec.cn/index.php/2017/12/14/ssd1306_oled_in_raspberry/

thijstriemstra, 2018-02-04:
No feedback, closing.

closed 2018-02-04 by thijstriemstra