rm-hull/luma.oled #163: Display issue with SSD1306 and SPI?

rogerdahl, 2017-08-11:
With my  SPI SSD1306 128x64 OLED display, the examples don't render correctly. [Here's a shot of maze.py.](https://user-images.githubusercontent.com/561416/29197762-71413c2c-7dfb-11e7-96e1-c17cbee73fac.jpg).

Has SPI with SSD1306 been tested with recent releases?

rm-hull, 2017-08-11:
Pasting your image here for easier reference

[image]

So I dont think it is an RGB issue: it is more likely that the init sequence (see https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L128-L135) is not quite right, the lower part of the image looks like it is not being written to, hence all the random dots.

I dont have an SPI version (only the I2C versions) - however I have a feeling that the SPI versions are mostly SH1106 devices, and if you look [here](https://github.com/rm-hull/luma.oled/blob/master/luma/oled/device.py#L54-L57) you'll see it has a different init sequence, and its memory is mapped differently.

Can you try and run the demo with `python examples/maze.py -d sh1106 --spi-port=... etc` instead to see if that solves it?

rogerdahl, 2017-08-11:
Thanks for the fast reply!

It works perfectly with `-d sh1106` :) I was focusing on that the display worked back when the library was called "ssd1306" but forgot that SH1106 support had already been added.

Would you like me to add this info and the screenshot to a troubleshooting section in the docs?

rm-hull, 2017-08-11:
Yes please, that would be great !

rogerdahl, 2017-08-13:
Done. I did a quick check on eBay but didn't find a good correlation between SPI / I2C  and SH1106 / SSD1306, so I just left that out of the troubleshooting note.

closed 2017-08-13 by rogerdahl

asavah, 2017-09-19:
Thanks.
I'll leave this here if anyone's google-fu brings them here.
http://www.waveshare.com/wiki/1.3inch_OLED_(B) is sh1106
while it's 0.96 inch counterparts is ssd1306